ros-industrial/kuka_experimental #27: rsi_hw_interface: improve example bringup launch files

gavanderhoorn, 2015-08-06:
The `kuka_rsi_hw_interface` package already includes a number of example launch files, but it would be nice to have a single set of examples that clearly shows (and documents) how to configure and use the `hw_interface` plugins / nodes within a typical `ros_control` setup.

Perhaps a good place for this would be a tutorial, but a set of accompanying launch files can be created in any case.

gavanderhoorn, 2016-06-27:
Some initial work in [gavanderhoorn/kuka_experimental/tree/kr6r900sixx_moveit_rsi_convenience](https://github.com/gavanderhoorn/kuka_experimental/tree/kr6r900sixx_moveit_rsi_convenience).

ankitagr92, 2017-10-06:
I need to interface Kuka KR60 JET with ROS but I did not find an suitable test code and instruction on how to Interface it.
Please guide me further.

gavanderhoorn, 2017-10-06:
Have you seen [kuka_rsi_hw_interface/krl/README.md](https://github.com/ros-industrial/kuka_experimental/blob/indigo-devel/kuka_rsi_hw_interface/krl/README.md)?

ankitagr92, 2017-10-06:
thanks..i will follow this once