ros-industrial/kuka_experimental #36: kr210_support: suspicious mass value in L150 xacro

gavanderhoorn, 2016-03-11:
The `inertial` element of the KR 210 L150 xacro macro contains a suspicious value for the `mass` of `base_link` ([here](https://github.com/ros-industrial/kuka_experimental/blob/105838a1edef85c300a83cd65b167f9592905a1c/kuka_kr210_support/urdf/kr210l150_macro.xacro#L12)). The base is probably heavy, but `1572.9` Kg seems a bit much.

@gomezc: did you set this value on purpose (to fix the base in place in simulation, fi)? Or is this a solidworks error?

gavanderhoorn, 2016-03-11:
Just noticed that all `link`s have rather high `mass`es specified. `link_1` is 1300 Kg, `link_2` still almost a 1000.

gomezc, 2016-03-11:
@gavanderhoorn I believe it was SolidWorks generated. I did not use robot in simulation, so I don't think that's why I set a mass value there. Looks like I got it from a step file, so it's likely a solidworks error. I can't seem to locate the original file to verify that those are the masses in SW.

gavanderhoorn, 2016-04-21:
Lacking any way (or time) to get more realistic values for the masses and inertias, I think I'm going to remove the `inertial` elements entirely. This will make it harder to use the model in Gazebo, but it is missing the required `dynamics` elements already, so that doesn't work right now anyway.

I'll leave this open for now, so if anyone watching has objections to this, please speak up. Otherwise I'll update the macro next week.

BrettHemes, 2018-05-22:
To be addressed via #124