ros-industrial/kuka_experimental #38: iiwa_stack + grl

ahundt, 2016-04-04:
If you aren't aware the library [iiwa_stack](https://github.com/SalvoVirga/iiwa_stack) (implemented at TUM) may be relevant to you. For high performance FRI drivers [grl](https://github.com/ahundt/grl) (implemented by me) may also be relevant.

Feel free to close this if desired, this is just an FYI.

gavanderhoorn, 2016-04-04:
Thanks for the pointers.

I've actually been aware of both of those for quite some time :). The `iiwa_stack` we've used here internally as well. The ROS additions to `grl` are a relatively recent addition, correct?

ahundt, 2016-04-05:
Yeah, we've implemented the ROS driver for a kuka innovation award project, I'll be out in germany for hannover messe with my group in a couple weeks.

ahundt, 2016-10-31:
We've since updated it to run with substantially higher performance in standard mode (udp+serialized flatbuffers) and it also has an FRI mode. What interface is used in [kuka_hardware_interface.cpp]( https://github.com/ros-industrial/kuka_experimental/blob/indigo-devel/kuka_rsi_hw_interface/src/kuka_hardware_interface.cpp)?

gavanderhoorn, 2016-10-31:
@ahundt wrote:
> What interface is used in [kuka_hardware_interface.cpp]( https://github.com/ros-industrial/kuka_experimental/blob/indigo-devel/kuka_rsi_hw_interface/src/kuka_hardware_interface.cpp)?

It's a node in `kuka_rsi_hw_interface`, so RSI.

Correct me if I'm wrong, but I have the impression that FRI is not something that is available on 'regular' industrial offerings of KUKA.

ahundt, 2016-11-01:
I think it is a set of software feature one can purchase with the arm. That's why I support two modes, one with FRI and one without.

ahundt, 2016-11-01:
sorry for the naive question, but what is RSI?

gavanderhoorn, 2016-11-01:
RSI is KUKA's Robot Sensor Interface. It's (not really, but still ..) a bit like FRIs smaller brother. 250 Hz position control, synced with controller. Connections over UDP or TCP. See [this page](http://www.kuka-robotics.com/en/products/software/hub_technologies/print/start.htm) (just a page I found) for KUKA's description. [KUKARobotSensorInterface.pdf](http://vip.gatech.edu/wiki/images/3/3c/KUKARobotSensorInterface.pdf) is a PDF manual of v2.3 (from 2009).

> I think it is a set of software feature one can purchase with the arm. That's why I support two modes, in grl one with FRI and one without, for example.

I know what FRI is :). I mainly meant that FRI is not something that's used with non-LWR robots.

kyjanond, 2016-11-02:
Just an addition: TCP/IP is not possible anymore for a newer versions of RSI (3.+) and it has actually two frequency modes (4ms cycle and 12ms cycle). FRI is an optional package for a Sunrise controllers and RSI for KRC4/KRC2, but as far as I know KUKA is planning to implement Sunrise also for KRC4 robots (or hopefully at least for the agilus line)

gavanderhoorn, 2016-11-02:
@kyjanond: thanks for the additions. That's certainly all true.

Sunrise for 'KR C4 robots' I've heard about as well. We'll have to see, I've heard quite some different opinions on the relative merits of the traditional KR Controllers and the Sunrise systems. But I guess that's partly personal as well (previous experience, understanding of modern programming language, etc).

ahundt, 2016-11-08:
cool, thanks for the info!