ros-industrial/kuka_experimental #39: Additional support for Kuka KR6

BrettHemes, 2016-04-20:
@gavanderhoorn  Continuing discussion from ros-industrial-consortium/descartes#124...

Yes, I just saw your kr6_support branch.  We can start work on contributing R700 sixx and R1100 sixx additions to compliment your existing R900 work.

Note: our visual meshes are currently all stl's; is this sufficient or should we work on trying to change them to be .dae's?

gavanderhoorn, 2016-04-21:
> Yes, I just saw your kr6_support branch. We can start work on contributing R700 sixx and R1100 sixx additions to compliment your existing R900 work.

Great.

> Note: our visual meshes are currently all stl's; is this sufficient or should we work on trying to change them to be .dae's?

Collada supports texturing and colouring, so if you'd like those models to have some detail to them, then yes as STL does't support that. Depending on how you created the meshes, exporting to Collada directly could preserve textures / colours, so that would not have to be any extra work. But otherwise STLs are fine. 

As for naming conventions and link frames you could use the current `kr6_support` pkg as a template. Just add the relevant bits for the R700 and R1100.

I'll also merge `kr6_support` into `hydro-devel`.

gavanderhoorn, 2016-04-21:
> I'll also merge `kr6_support` into `hydro-devel`.

done.

gavanderhoorn, 2016-04-21:
Note btw that the branch is still called `hydro-devel`, but that is only because things are still compatible with that. Everything can be used in newer releases as well.

gavanderhoorn, 2016-06-16:
@BrettHemes: are you still interested in contributing the other Agilus variants?

BrettHemes, 2016-06-17:
Yes, indeed; we have figured out the submission approval process from our
end so now it's just a matter of finishing it.  Did you have an immediate
need timeframe for finishing the package?  We have been focusing on other
stuff but can work on wrapping this up. I have kr6r700sixx done.  Do you
care if we submit single variants at a time?  Might make sense to see if
our style/namings are good before doing everything.

@BrettHemes <https://github.com/BrettHemes>: are you still interested in
contributing the other Agilus variants?

gavanderhoorn, 2016-06-17:
> Did you have an immediate need timeframe for finishing the package?

No, not really. I was just curious if you were still interested, as it'd been a while since you opened the ticket.

> Do you care if we submit single variants at a time?

No, that's perfect actually. I'd prefer that.

> Might make sense to see if our style/namings are good before doing everything

Exactly.

BrettHemes, 2016-06-23:
@gavanderhoorn Do you want moveit configs and/or ikfast solutions too or just the meshes and xacros?

gavanderhoorn, 2016-06-23:
Let's do the support pkg additions first. The other pkgs are interesting too, but depend on the support pkg.

gavanderhoorn, 2016-06-23:
Btw, I've added an `indigo-devel` branch. Be sure to submit the PR against that.

cschindlbeck, 2016-07-02:
@BrettHemes you mentioned that you might work on r1100 sixx cad data. You mean kr10 r1100, right? Did you finish that? I have a kuka kr10r1100 and might work on the corresponding cad data in the coming days /weeks if you haven't.

BrettHemes, 2016-07-02:
Yeah go ahead,  haven't made much progress on that front yet.  Thanks!

BrettHemes, 2016-07-11:
KR6 R700 files added via PR #58

gavanderhoorn, 2016-07-12:
@cschindlbeck: it would be very much appreciated if you could work on the KR 10 R1100. If you decide to do that, you could use the files for the R900 as a template.

cschindlbeck, 2016-07-12:
I will definitely do that in the upcoming days or weeks when I have a bit more time. As you already pointed out probably only link 2 and 4 are different from the other models with regard to cad and kinematics.

cschindlbeck, 2016-07-15:
So....i tried to export the meshes for ROS and i encounter some difficulties
My approach (e.g. link 4):
- Open link in Solidworks
- Add appropriate coordinate system to link
- export as .stl file
- open in meshlab to fully close the meshes
- open in blender with DAE plugin to export meshes for "visual"

My problem is when i open the .stl file in blender/rViz the link is HUGE and the size is not consistent.
Somehow the size is not preserved after exporting via Solidworks

(rant: the CAD model provided by KUKA is crap!)

gavanderhoorn, 2016-07-15:
> My problem is when i open the .stl file in blender/rViz the link is HUGE and the size is not consistent.

ROS uses meters for everything. SolidWorks by default exports in mm (for STL at least). Make sure you set the units to meters in the STL save dialogue in SolidWorks.

gavanderhoorn, 2016-07-15:
Also: choose the high quality save option. You always decimate the mesh in Blender afterwards if needed, and I've always found the normal quality to be insufficient for highly curved geometry.

Btw: it only makes sense to convert to Collada if you are going to texture or colour faces/vertices. As you already only have STLs, just converting to Collada by itself is not going to add anything.

gavanderhoorn, 2016-07-15:
And finally: the KR 10 series will need its own support package (`kuka_kr10_support`). We cannot store the files in the KR 6 package.

Just copy `kuka_kr6_support` and search-replace `kr6 -> kr10`.

gavanderhoorn, 2016-08-30:
Just an update (although slightly off-topic in this issue): I just merged the KR10 support package contributed by @cschindlbeck. See #64.

gavanderhoorn, 2016-08-30:
Closing. @BrettHemes has contributed the R700 in #58.

Thanks again @BrettHemes.

closed 2016-08-30 by gavanderhoorn