ros-industrial/kuka_experimental #57: rsi_hw_iface: semantics of 'rsi/address' and 'rsi/port' params unclear

gavanderhoorn, 2016-07-11:
Perhaps only for me, but because of the structure of the `rsi_hw_interface` node, `rsi/address` and `rsi/port` configure the address and port combination to which the UDP socket created by `UDPServer` is bound, and thus should correspond to an ip local to the ROS host running the `rsi_hw_interface` node.

gavanderhoorn, 2016-07-15:
I suggest changing these to `rsi/local_address` and `rsi/local_port`.

We could perhaps include `ros` somewhere, but I'm not sure I like `rsi/ros_address` and `rsi/ros_port` too much. Maybe `rsi/address_ros` and `rsi/port_ros`.

@tingelst, @pgorczak. @bj0rkedal ?

tingelst, 2016-07-15:
I suggest changing these to rsi/local_address and rsi/local_port.

I agree.

bj0rkedal, 2016-07-16:
Agreed, `rsi/local_address` and `rsi/local_port` is clearer.

gavanderhoorn, 2016-07-16:
Another suggestion (and I like this best):
```yaml
rsi:
   listen_address: "1.2.3.4"
   listen_port: 49152
```

This more clearly conveys the fact that the node uses these to determine on which IP address and port it should listen for incoming RSI traffic from the controller.

gavanderhoorn, 2016-07-16:
See #63 for a PR that renames the parameters to `listen_address` and `listen_port`.

gavanderhoorn, 2016-07-18:
@tingelst @bj0rkedal: any comments?

bj0rkedal, 2016-07-18:
In my opinion it's a good improvement. I think this should make it clear to any user what's configured. With additional input from the expanded documentation it should not take long to get a working system.

BrettHemes, 2016-08-19:
Personally I find this much clear and vote for the listen_address/port naming (I had renamed them to rsi_listener_ip/port in my launch files).

gavanderhoorn, 2016-08-30:
Fixed in f9224e49479d7fcd5c1ac3d15ce0ef6938fe81ee.

closed 2016-08-30 by gavanderhoorn