ros-industrial/kuka_experimental #62: kr6r900sixx: moveit planner problems (L)BKPIECE

bj0rkedal, 2016-07-15:
I have not managed to get BKPIECEkConfigDefault, LBKPIECEkConfigDefault or LBKPIECE1 to plan any sensible trajectories in my moveit config based on the current kr6 support package. All other available planners works, except TRRTkConfigDefault which returns an assertion error.

Is this a problem only on my end?

gavanderhoorn, 2016-07-15:
Those planners have been problematic with other robots as well. See ros-industrial/universal_robot#193 fi (and related issues like ros-industrial/fanuc#156).

Not sure what is going on precisely, but for now a work-around is to just use `RRTConnect` in these cases.

For the pkgs in `ros-industrial/fanuc` we've made it the default, see ros-industrial/fanuc#188.

gavanderhoorn, 2016-07-15:
I've also made it the default for the moveit config in `kr6r900sixx_moveit_rsi_convenience` ([here](https://github.com/gavanderhoorn/kuka_experimental/commit/6c1c096f9c191671d081315f40d75c1b08905fc6)).

bj0rkedal, 2016-07-15:
Thanks for clearing that up.

closed 2016-07-15 by bj0rkedal