ros-industrial/kuka_experimental #65: Kinematics of Agilus

cschindlbeck, 2016-07-20:
I noticed when you align the robot with respect to the zero markers (mastering) that the robot is not in a "geometric" zero position. Right now, in the kuka_experimental package, the kr6 and kr10 variants are all zero in a geometric position, which means that specifying (0,0,0,0,0,0) does not exactly correspond to the CAD model. The problem is that the robot is slightly asymmetric.

Maybe i am also mistaken but i think we are off about 1-3°. 

Here are pictures displaying the issue:

[image]
[image]

gavanderhoorn, 2016-07-20:
Hm. That does seem strange indeed, and not something I've noticed before.

Would it be an option for you @cschindlbeck to contact your KUKA office about this? I think they would be the ones able to provide us with a definitive explanation for this.

bj0rkedal, 2016-07-21:
I've also noticed this misalignment of the CAD model zero markers. However, I moved a real KR6 R900 to the joint angles corresponding to the CAD pose (0,-90,90,0,0,0) and the results are the same. I have not calibrated the joints myself, but they should have been correctly calibrated by someone else some time ago.

It should still be confirmed if this is wrong or not.
[image]

tingelst, 2016-07-21:
These marks are the premastering marks, that is, marks that specify the initial joint positions the robot should be placed in before the mastering (joint zero calibration). The mastering is performed automatically using an electromechanical device. 

The CAD model is thus correct.

gavanderhoorn, 2016-07-21:
Thanks for the clarification @tingelst.

@cschindlbeck: I think you changed the meshes in d427d60d3cde57c6909475ac81a45ec7597a0f4f to correct for this? It would seem those changes were unnecessary. Would you agree they should be reverted?

cschindlbeck, 2016-07-21:
No the opposite is the case, the CAD model now looks like the pictures above when put in zero position. d427d60d3cde57c6909475ac81a45ec7597a0f4f should be fine, however, someone should check that before merging to main repo just to be safe. 

If premastering marks do not correspond to real zero position i can safely close this thread, thanks for clearing that up.

closed 2016-07-21 by cschindlbeck