ros-industrial/kuka_experimental #67: Kuka Kr6r900sixx Joints not found

mat-l, 2016-08-09:
Hey all,
I tried to use the Kr6r900 in MoveIt. Therefore I created an kr6_moveit_config folder with the moveit_assistant using the xacro_urdf file from /kuka_kr6_support/urdf folder. After navigating to the launch folder within the moveit_config folder I started demo.launch in Terminal.
As result RViz opens with the Robot but if I try to plan a random path it consumes a lot of time and the Solution found is a complicated Path with lots of useless motions. In Terminal I get the following Error for each of the Joints:

[ERROR] [1470755272.462600829, 8008.567000000]: Joint 'wrist_2_joint' not found in model 'kuka_kr6r900sixx'

It seems like MoveIt is trying to load the Joint from the Kr6r900sixx file but instead of searching for Joint 1 to Joint 6 it uses the names from the UR Package. 
At first I had the UR and Kuka files in the same Catkin Workspace. But even after seperating them in two different Workspaces the Error occured.
[image]
[image]
[image]
[image]
[image]
[image]
[image]

[image]

Any suggestions are appreciated. 
Thanks

shaun-edwards, 2016-08-09:
Have you looked through the `yaml` files that the setup assistant generated?  There should be a list of the joint names in one of them (I believe it's `*.srdf`).

mat-l, 2016-08-09:
Thank you for your super fast answer.
[kuka_kr6_moveit_config.zip](https://github.com/ros-industrial/kuka_experimental/files/409301/kuka_kr6_moveit_config.zip)
This is the config file that has been created. I checked them but they seem to be fine.

gavanderhoorn, 2016-08-09:
A few things:

 1. specify either a chain or a group of joints, not both (I'd recommend a chain, from `base_link` to `tool0`)
 1. unless you have a specific need / setup, don't make `odom_combined` the parent of your fixed frame
 1. make sure you don't / didn't have a separate instance of `roscore` running in some other tab / session (that could persist parameters that are not overwritten by `demo.launch`, which we don't want)

> As result RViz opens with the Robot but if I try to plan a random path it consumes a lot of time and the Solution found is a complicated Path with lots of useless motions

Make sure you have selected a good planner on the MoveIt planning request setup tab. See #62.

> I tried to use the Kr6r900 in MoveIt. Therefore I created an kr6_moveit_config folder with the moveit_assistant using the xacro_urdf file from /kuka_kr6_support/urdf folder.

Please see if [gavanderhoorn/kuka_experimental/kr6r900sixx_moveit_rsi_convenience](https://github.com/gavanderhoorn/kuka_experimental/tree/kr6r900sixx_moveit_rsi_convenience) works for you. That branch contains a KR6 R900 sixx MoveIt configuration with everything setup to use it with ROS-Industrial (or just through the `demo.launch` file).

If you plan on using this with the `kuka_rsi_hw_interface` package and a real robot, please make sure to replace [this line](https://github.com/gavanderhoorn/kuka_experimental/blob/6d564a98842cf5f9c871190f06b7e3cbef351134/kuka_rsi_hw_interface/launch/robot_interface_streaming.launch#L54) with the IP address of your ROS PC that it uses to communicate with the robot, and to read the updated RSI setup instructions in #54 (direct link to the readme: [here](https://github.com/bj0rkedal/kuka_experimental/blob/update_readme/kuka_rsi_hw_interface/krl/README.md)). Experiences of a KR10 R1100 user at #56.

gavanderhoorn, 2016-08-09:
As to your original problem: I suspect this is something local to your machine / ROS environment setup, rather than an issue with the xacro we have in the `kr6_support` package.

As you noted yourself, the `kr6r900sixx` xacro doesn't contain those joints, so it would seem strange for them to show up in a MoveIt configuration based on it.

mat-l, 2016-08-11:
Even though I could not solve the original Problem you're MoveIt setup works perfectly fine for me.
I plan to use a real Robot later so the instruction is really helpful as well.
Thank you for that!
The next step would be to connect MoveIt with a simulated kr6 in Gazebo. You wouldn't have a Gazebo package ready for kr6r900sixx, would you?

gavanderhoorn, 2016-08-14:
> The next step would be to connect MoveIt with a simulated kr6 in Gazebo. You wouldn't have a Gazebo package ready for kr6r900sixx, would you?

No, I'm afraid I don't.

gavanderhoorn, 2016-08-14:
@mat-l: I'm going to close this issue, as I believe you have found a working solution (although it isn't an actualy *solution* to your original issue).

If you want to continue troubleshooting, please re-open and provide us with more information and a way to reproduce the problem.

closed 2016-08-14 by gavanderhoorn