ros-industrial/kuka_experimental #69: Exception when sending joint trajectories to kuka_rsi_hw_interface FollowJointTrajectoryAction interface

BrettHemes, 2016-08-24:
@gavanderhoorn @tingelst @bj0rkedal

I am having trouble getting Descartes-generated trajectories to run using the rsi interface.  On an SimpleActionClient::sendGoal(...) call I get the following exception:

> [ERROR] [1472046592.407557364]: Unexpected exception caught when initializing trajectory from ROS message data.

Seems to be from this line: http://docs.ros.org/indigo/api/joint_trajectory_controller/html/c++/joint__trajectory__controller__impl_8h_source.html#l00518

After throwing, the action server reports success but the robot doesn't move at all...

I figure I am doing something silly, but am confused by the fact that the **same** code runs as expected using the `industrial_robot_simulator` in place of `kuka_rsi_hw_interface` (which is why I posted here).

Aslo, the issue seems to only be with my (Descartes) trajectories, as anything planned/executed using the movegroup interface executes without issue.

BrettHemes, 2016-08-24:
Nevermind; I fixed this.  

I did some more digging and found some additional documentation [here](http://docs.ros.org/hydro/api/joint_trajectory_controller/html/c++/namespacejoint__trajectory__controller.html#a8b55b5e436dc0e49ad024d2f3bd7568c) that provided some insight.  I ended up being that my trajectory's header time stamp was too far in the past and was causing the issue.

I guess the `industrial_robot_simulator` doesn't care about this.

closed 2016-08-24 by BrettHemes

gavanderhoorn, 2016-08-25:
Hm. I was going to suggest looking at the trajectories themselves, but seems like you already did that.

> After catching, the action server reports success but the robot doesn't move at all...

It would be nice if the `joint_trajectory_controller` emitted a bit more helpful error message. Not sure if that is possible at that point, but could you at least report this to the `ros_control` people?

> I guess the `industrial_robot_simulator` doesn't care about this.

But it should: could you report this please to `industrial_core`?