ros-industrial/kuka_experimental #70: rsi_hw_iface: node shuts down on agressive motions

gavanderhoorn, 2016-09-03:
As reported by @BrettHemes in https://github.com/ros-industrial/kuka_experimental/pull/54#issuecomment-240841929:

> I was able to trigger `hw_iface` shutdowns while in AUTO using the RQT Joint Traj Controller by somewhat aggressively dragging a slider.  Could be a priority issue or a robot torque limit as mentioned above (running non-real-time) but I just wanted to note that you can reproduce cschindlbeck's issue (although not as easily) outside of T1.

gavanderhoorn, 2016-09-03:
@BrettHemes: could clarify a bit more what you experienced? We might want to add some additional safety-checks / limiting to avoid these kind of things.

BrettHemes, 2017-02-01:
This appears to be a fault thrown from within the KRC controller for trajectories that violate the internal limits (joint accel, error, etc.).  With the RQT Joint Traj Controller it is possible to move the sliders faster than the robot can react, thus causing a fault.  I don't really have an idea for how to address this (or if we even should).  At the moment, I think requiring feasible trajectories from the user is appropriate.

Anything generated from MoveIt! with correctly specified joint limits shouldn't cause this issue.

BrettHemes, 2017-11-16:
Added a note regarding this to the troubleshooting guide [here](http://wiki.ros.org/kuka_rsi_hw_interface/Troubleshooting)

gavanderhoorn, 2017-11-16:
Thanks, much appreciated. 💯

closed 2017-11-16 by gavanderhoorn