ros-industrial/kuka_experimental #71: rsi_hw_iface: clarify how the RSI 'timeout' parameter works exactly

gavanderhoorn, 2016-09-03:
From https://github.com/ros-industrial/kuka_experimental/pull/54#issuecomment-240841929:
> Is the timeout equal to the number of allowed missed packets? So at Timeout=100, 100*4ms have elapsed since the last answered RSI packet?

The text does not really explicitly state it like this, but assumes that the reader is able to understand it from the explanations further on in the same paragraph.

We might want to add a small example calculation that shows how `timeout` and actual time are related. Perhaps mention that timeout is measured in 'control periods'?

BrettHemes, 2017-02-01:
I believe this was sufficiently addressed in #54 

> +Notice the RSIObject of type `ETHERNET`. Within this object is a parameter called **Timeout**. This parameter is set to **100** by default. The RSI interface operates at `250 Hz` (4ms cycle). The **kuka_rsi_hardware_interface** does not have a period configured and is completely driven by the controller's output. Every controller RSI output has a IPOC timestamp which increments for every cycle. The **kuka_rsi_hardware_interface** will answer as quickly as possible. The answer includes the last IPOC received. If the connected **PC with ROS** did not manage to answer within the RSI cycle of **4ms**, the IPOC timestamp of RSI has incremented. The IPOC included in the answer is not matched and the controller will increment a counter. When this counter hits the **Timeout** parameter (**100**), the RSI connection will shut down. If this parameter is lowered, the demand for real-time computation will increase.

BrettHemes, 2017-11-16:
Documented on the troubleshooting guide via the wiki [here](http://wiki.ros.org/kuka_rsi_hw_interface/Troubleshooting)

gavanderhoorn, 2017-11-16:
👍 thanks again.

closed 2017-11-16 by gavanderhoorn