ros-industrial/kuka_experimental #73: rsi_hw_iface: Robot won't connect with fresh install of Indigo

BrettHemes, 2016-09-19:
I can not seem to get the RSI HW Interface working on a fresh install of Indigo.  I have tried on two new machines (one a laptop and one a desktop) so either I am missing a step at this point or something on ROS's side changed.  I have pushed all of my RSI-related files from the working setup to a git repo and then pulled them to the new machines to ensure all the configs are identical.  **Nothing on the Kuka controller has been changed.**  

From a fresh Ubuntu 14.04.4 install I did:

1. install ros-indigo-desktop-full
2. install ros-indigo-ros-control
3. install ros-indigo-ros-controllers
4. checkout kuka_experimental (package copied from good/working system)
5. catkin_make -> all builds successful
6. roslaunch kuka_rsi_hw_interface test_hardware_interface.launch sim:=false

after which I get the following output:
```
started roslaunch server http://jake:36901/

SUMMARY
========

PARAMETERS
 * /controller_joint_names: ['joint_a1', 'joi...
 * /joint_state_controller/publish_rate: 50
 * /joint_state_controller/type: joint_state_contr...
 * /position_trajectory_controller/action_monitor_rate: 20
 * /position_trajectory_controller/joints: ['joint_a1', 'joi...
 * /position_trajectory_controller/state_publish_rate: 50
 * /position_trajectory_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.20
 * /rsi/address: 192.168.2.100
 * /rsi/port: 49152

NODES
  /
    controller_spawner (controller_manager/spawner)
    kuka_hardware_interface (kuka_rsi_hw_interface/kuka_hardware_interface_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [23639]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 64d0444a-7e76-11e6-95f1-18a6f7016e94
process[rosout-1]: started with pid [23652]
started core service [/rosout]
process[kuka_hardware_interface-2]: started with pid [23656]
process[controller_spawner-3]: started with pid [23660]
process[robot_state_publisher-4]: started with pid [23663]
[ INFO] [1474295765.500010363]: Starting hardware interface...
[ INFO] [1474295765.510205895]: Loaded kuka_rsi_hardware_interface
[ INFO] [1474295765.511812716]: Setting up RSI server on: (192.168.2.100, 49152)
[ INFO] [1474295765.541849040]: Waiting for robot!
[INFO] [WallTime: 1474295765.960549] Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [WallTime: 1474295765.963120] Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [WallTime: 1474295765.965221] Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [WallTime: 1474295765.967296] Loading controller: position_trajectory_controller
[INFO] [WallTime: 1474295766.056052] Loading controller: joint_state_controller
[INFO] [WallTime: 1474295766.098614] Controller Spawner: Loaded controllers: position_trajectory_controller, joint_state_controller

```

... and then I start the Kuka program only to get the error
`Object ETHERNET11 returns error RSITimeout`

I am stumped at this point as I can put the Ethernet cable in the good machine (with the same IP address) and hit play on the Kuka and it works....

One difference I did notice is that the non-working setups report /rosversion: 1.11.20 on launch while the good/working setup reports /rosversion:1.11.19   I would try and do dist-upgrade to see if that is indeed the issue but I am scared of breaking everything...

Could someone possibly confirm that I am not missing something obvious and/or offer any suggestions on how to figure this out?

gavanderhoorn, 2016-09-19:
If the new machine doesn't have the same IP as the old one, have you made sure to update the configuration on the robot as well?

BrettHemes, 2016-09-19:
They have the same IP.

gavanderhoorn, 2016-09-19:
Can you verify with Wireshark that the robot is actually trying to send UDP packets?

BrettHemes, 2016-09-19:
I did an apt-get dist-upgrade on the working machine and rebuilt my workspace; it still works... so the problem is on my side some where.

Running Wireshark, I do indeed see packets from the robot.  I get about 150 of them before the timeout.  Maybe the default RSI timeout is triggering?

BrettHemes, 2016-09-19:
I changed the Timeout to 200 and see exactly 201 packets from the robot in Wireshark before the error.  Nothing seems to be going from the PC back to the robot.

BrettHemes, 2016-09-19:
Yeah, whatever I set the timeout to, I get exactly one more packet than that from the robot....

BrettHemes, 2016-09-19:
The hw_iface is stuck in this call
[https://github.com/ros-industrial/kuka_experimental/blob/indigo-devel/kuka_rsi_hw_interface/include/kuka_rsi_hw_interface/udp_server.h#L152](https://github.com/ros-industrial/kuka_experimental/blob/indigo-devel/kuka_rsi_hw_interface/include/kuka_rsi_hw_interface/udp_server.h#L152)

gavanderhoorn, 2016-09-19:
Are you really sure that the IP you configure is the same? Netmask as well? The fact that the code waits in `recvfrom(..)` basically implies that no data is being received on the opened socket.

No firewalls enabled in the new Ubuntu installation?

tingelst, 2016-09-20:
We had some similar experience some months ago. I was able to get the system up, but I am not 100 % sure of the solution. Try to use the python script found [here](https://gist.github.com/tingelst/1ef24d4650e208a94d2080fa3f6795cd) and see if you can receive any data on the socket.

Usage:
```bash
python udp_printer.py --host=192.168.2.100 --port=49152
```

gavanderhoorn, 2016-09-20:
In case you have it on your system, `ncat -klu 49152` should also work (part of the `nmap` deb pkg).

BrettHemes, 2016-09-20:
Ok, I figured it out.  While not a kuka_hw_iface issue directly, it is related to the networking / RSI interface on the controller...

I am by no means an expert when it comes to networking but while googling networking issue topics last night, I came across some comments to the effect that some networking hardware can store MAC address for a particular IP and thought that whatever is in the controller itself might be doing something along these lines.  With that in mind, I forced a cold start and rebooted the controller and the problem was resolved.  Just to note, I had tried to power cycle early in the debugging process but apparently the **hibernate setting in the controller is not enough to fix the issue**.  

I confirmed the issue/fix by trying with the previously good setup (with the same IP address) and after a cold start and successful connection by the new PC, was able to reproduce the above.

**So to summarize, the RSI interface will only work with the first connected PC for a given IP until the next cold start of the controller.**  This is only an issue when connecting a new PC to the rsi interface (X66) with the **SAME** IP address as a previously used PC...   that said, simply changing the IP address (IP_NUMBER tag in ros_rsi_ethernet.xml) for any new hardware also solves the issue without requiring a reboot (tested).

To force the cold start I did:
```
Main Menu > Shutdown > Settings that are only valid next time the system is switched off: Force cold start
```
followed by a power cycle on the controller.

tingelst, 2016-09-20:
That's great to hear!

gavanderhoorn, 2016-09-20:
Thanks @BrettHemes for reporting back, and for figuring this out. ARP caching can indeed be an issue, but not one I assumed would be responsible for this.

@tingelst: we might want to document these kind of things somewhere. Perhaps a *Troubleshooting* page on the wiki (similar to [fanuc/Troubleshooting](http://wiki.ros.org/fanuc_driver/Troubleshooting)).

BrettHemes, 2016-10-08:
Feel free to close this as your see fit

gavanderhoorn, 2016-10-28:
Thanks @BrettHemes.

I've captured the need to document these kind of things in #75.

closed 2016-10-28 by gavanderhoorn