ros-industrial/kuka_experimental #78: wrong parent and child in lbr_iiwa_14_r820.urdf

ommmid, 2017-04-20:
urdf_to_graphiz of this file:  lbr_iiwa_14_r820_macro.xacro can be found in the following link:
https://drive.google.com/file/d/0B4AVR3vrx7i3TGo2WDFYMGx3WTA/view?usp=sharing
base_link is the parent of base which I think is wrong. The parent should be ${prefix}base and the child should be ${prefix}base_link. They are in reverse in the code. Am I right?

    <link name="${prefix}base" />
    <joint name="${prefix}base_link-base" type="fixed">
      <origin xyz="0 0 0" rpy="0 0 0"/>
      <parent link="${prefix}base_link"/>
      <child link="${prefix}base"/>
    </joint>

gavanderhoorn, 2017-04-25:
Thanks for reaching out.

I believe this is actually correct. See [](http://wiki.ros.org/Industrial/Tutorials/Create%20a%20URDF%20for%20an%20Industrial%20Robot#Additional.2BAC8-Standard_Frames) for the definition of the `base` frame.

`base_link` is the *de facto* start of all URDF models in ROS. We added `base` (and `tool0`) to be able to ask TF for the transform between these two frames which should correspond to whatever the robot controller reports as the Cartesian pose of its TCP. There is no requirement for `base` to be part of the main kinematic chain, but it does need to be connected to it via some transform.

> The parent should be ${prefix}base and the child should be ${prefix}base_link.

Could you explain why you feel that would be necessary?

ommmid, 2017-04-26:
Thanks for the reply.
About the name by ROS convention, you are right. But my question is why we need the second branch: [base_link] -> (base_link-base) -> [base] while this Kuka robot is just a serial arm having all the joints and links in one chain. Then the chain: [base_link] -> (joint_a1) -> [link_1] -> ... could work for the transformations in TF. base_link is the first link at 0,0,0 and all the other links in series up to tool0 which seems fine. Why do we need the second chain?

gavanderhoorn, 2017-04-26:
> Why do we need the second chain?

we don't necessarily *need* it, but it seemed like the most convenient (and efficient) way to attach the `base` frame to the rest of the model.

`base` was introduced after we already had a large nr of robots modelled in ROS-Industrial. Inserting `base` into the kinematic chains that were already there was a potentially breaking change, so we opted to not do it that way.

Secondly: in many industrial robot controllers, the origin of the *world frame* is at a point defined by the first and second axes, but it's orientation is fixed in some manner (ie: it does not change when `joint_a1` changes its orientation). To implement that - while making `base` part of the main chain - we'd have to do some trickery with the origin of `link_1` or `link_2` and add (an) additional frame(s) (not necessarily with KUKA robots, but the concept is used in all ROS-Industrial robot support packages). Making `base_link` the parent of `base` directly avoids all of that.

Finally, right now `base` is like an additional frame that provides some extra semantics to the model, in a separate "layer" almost. They can be taken out / supressed without too many edits to the xacro macro. If they were part of the main chain that would be more involved.

In conclusion: `base` could be part of the main chain but implementation and modelling considerations have led us to not do that.

ommmid, 2017-04-26:
Thanks for the detailed answer. I think I get it now.

gavanderhoorn, 2017-04-26:
No problem.

I'm closing this then. Feel free to keep commenting and / or re-open the issue if you see a need.

closed 2017-04-26 by gavanderhoorn