ros-industrial/kuka_experimental #81: kuka krl_file_parser support

yijiangh, 2017-05-11:
Based on comment from Hemes in [google group discussion](https://groups.google.com/forum/#!topic/swri-ros-pkg-dev/0iV4l6de3I0)[2], I'm proposing the development of a **krl_file_parser** support here, enabling translating generated trajectory from motion planner in moveit architecture (OMPL or descartes) into kuka-executable [KRL](https://en.wikipedia.org/wiki/KUKA_Robot_Language)[1] file, so that we can execute the generated robot motion by copying and pasting KRL file (via usb or direct upload) to KRC controller and execute them by T1 (hand-held safe mode for kuka) or Auto mode. 

For application without the need of sensoring and live-feedback, I think this might be a good and safe way to unleash the power of  ros and moveit! to broader industrial robot user and researcher community, as RSI and MxAutomation [issue #80] are often vendor-stingy and cost-intensive ([check Hemes's comment on May-4-17](https://groups.google.com/forum/#!topic/swri-ros-pkg-dev/0iV4l6de3I0))[1]. And I assume for industrial users with the worry of having different trajectories every time they use Moveit OMPL for same target and goal (check the [talk](https://vimeo.com/142629935)(21'33'') from Shaun Edwards at ROSCon2015 [3]) can be mitigated by using this method. Users can save their moveit-generated motion from offline planning in ros, without having to setup connection between their ros-server and real robot controller (and expensive controller interface controller as well), allowing them to preform the same moveit trajectory for specific goal every time they run it using the file. This parser might be good for simple motion demonstration without live-feedback, and also allow users to exploit ros moveit's power with no intention or condition for having required vendor-based software and hardware.

For [KRL](https://en.wikipedia.org/wiki/KUKA_Robot_Language) [1], a very comprehensive (a bit old, butr still working really well) manual can be found here[4].

Happy to hear opinions and suggestions.

Best,
Yijiang

[1] https://en.wikipedia.org/wiki/KUKA_Robot_Language
[2] https://groups.google.com/forum/#!topic/swri-ros-pkg-dev/0iV4l6de3I0
[3] https://vimeo.com/142629935
[4] forum.robotsinarchitecture.org/index.php?action=dlattach;topic=20.0;attach=11

gavanderhoorn, 2017-06-12:
Hi @yijiangh, thanks for opening this issue here.

First: I believe what you are describing / asking for would be a post processor or code generator instead of a parser. The former would *produce* KRL to send it to a controller, while the latter would *consume* it for some purpose.

That aside, having a post processor targeting KUKA would definitely be of interest, and I believe we have a good starting point over at [ros-industrial/robodk_postprocessors](https://github.com/ros-industrial/robodk_postprocessors). The developers of RoboDK have contributed all the post processors available in RoboDK and have been very responsive to suggestions and bug reports.

The idea is to wrap the post-processors in a ROS node that exposes appropriate services and actions to make it possible to transform generic ROS representations for trajectories into controller-specific programs. I think you've already found some discussion about that over at ros-industrial/robodk_postprocessors#1.

gavanderhoorn, 2017-06-12:
A few notes about some specific things.

> And I assume for industrial users with the worry of having different trajectories every time they use Moveit OMPL for same target and goal (check the [talk](https://vimeo.com/142629935)(21'33'') from Shaun Edwards at ROSCon2015 [3]) can be mitigated by using this method.

This is definitely an approach to improve determinism and predictability. An alternative could be to exploit a caching system between MoveIt and planning consumers that could remove the need to replan for every request completely, provided start and end states (and other relevant pieces of the robot's state) are identical / close enough to a stored one.

> Users can save their moveit-generated motion from offline planning in ros, without having to setup connection between their ros-server and real robot controller (and expensive controller interface [...]

Very important here is to note that you obviously lose the replanning and obstacle avoidance capabilities whenever re-used plans are used.

gavanderhoorn, 2017-06-12:
@yijiangh: would you be ok with centralising the discussion about code generation/post processors, and close this issue in favour of ros-industrial/robodk_postprocessors#1?

yijiangh, 2017-06-12:
Hi @gavanderhoorn : Thanks for your reply and discussion. 

I wrote this commit before I joined the discussion in [robodk_postprocessor](https://github.com/ros-industrial/robodk_postprocessors). Sure I will close this issue and recommend whoever shares the interest in this to move to robodk_postprocessors.

closed 2017-06-12 by yijiangh