ros-industrial/kuka_experimental #84: kuka_rsi_hw_i : Failed to read state from robot. Shutting down!

SNBasti, 2017-05-19:
# Error Description

I am able to start the RSI communication with the Controller and the ROS PC. 
However after launching the gui to control the joints ( `rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller` ) e.g. to move joint A1 Things more or less immediately stop working ( or at least after a few seconds ).  
The `kuka_rsi_hw_interface` node stops working producing the following error: 

```
[FATAL] [1495202179.132217349]: Failed to read state from robot. Shutting down!
================================================================================REQUIRED process [kuka_hardware_interface-1] has died!
process has finished cleanly
log file: /home/sebtut/.ros/log/d7c77dc8-3c9a-11e7-8f09-c85b760307ba/kuka_hardware_interface-1*.log
Initiating shutdown!
================================================================================
```

The `SmartHMI` produces: 

```
Block change on STOP
Originator: /R1/ROS_RSI  

Stop by $CORRECTION function ( reset or block selection required) 

Command velocity A1 
```

# ROS Installation 

I installed ROS on the following machine 

```
uname -a 
4.4.0-53-generic #74-Ubuntu SMP Fri Dec 2 15:59:10 UTC 2016 x86_64 GNU/Linux
```

I installed ROS kinetic using `apt-get`.  
The `ros industrial core package` and the `ros kuka experimantel package` I build myself by cloning them into a catkin workspace and running `catkin_make`. Additionally I had to install the `ros joint_trajectory_controller` ( http://wiki.ros.org/joint_trajectory_controller ) using `apt-get` to get the ros_rsi_hw_i` node running. 

# KUKA Hardware

I use the KUKA KR 6 R900 sixx together with KRC4 controller. 
In the launch file I added 
```
<param name="robot_description" command="$(find xacro)/xacro.py '$(find kuka_kr6_support)/urdf/kr6r900sixx.xacro'"/>
```

AAvi, 2017-05-19:
Did you try installing RT-Preempt?
It worked for me with the same controller. Also set priority of RVIZ to very low. 

"If you have problems with the connection to RSI shutting down now and then while moving the robot it is suggested to:

Compile and install a RT-Preempt kernel for your PC.
Give kuka_rsi_hardware_interface on your PC real-time priority when the RSI connection is established.
"

SNBasti, 2017-05-22:
Thanks for the hint. I already considered that this may is the problem. 
Will test it this week and update the issue.

SNBasti, 2017-05-26:
Patching the kernel as PREEMTIVE made it stable

closed 2017-05-26 by SNBasti

maxgittelman, 2017-09-07:
I understand that I should give kuka_rsi_hardware_interface real-time priority, and that I should patch the kernel as PREEMPTIVE but there doesn't seem to be any guide as to how to do this? 

I found an [installation here](https://wiki.linuxfoundation.org/realtime/documentation/howto/applications/preemptrt_setup) and this [command here](https://rt.wiki.kernel.org/index.php/Frequently_Asked_Questions) `CONFIG_PREEMPT_RT_FULL=y` . But there is no reference as to how to specifically give the kuka_rsi_hw_interface priority.

Any tips on how to do this, should I be adding a line to the `kuka_hardware_interface_node.cpp`?

destogl, 2017-09-08:
There is a lot of example in Internet. you need just install preempt kernel from your repositories....

SebNag, 2017-09-08:
If you are running `Ubuntu` it may be sufficient to install a _soft realtime low latency kernel_ from the standard repositories. It's a simple `sudo apt install` command then. 

I wrote some notes on this issue, you can view them [here](https://github.com/SebNag/Personal_Wiki/blob/master/Linux/LinuxKernel/Patches/info.md).

maxgittelman, 2017-09-08:
So I figured out how to use `nice` and `renice` to update the priority of the `kuka_hardware_interface` process to the highest priority of "-20" but upon sending a JointTrajectoryGoal to hardware I get the following error message from the terminal running the `kuka_hardware_interface`:
```

[FATAL] [1504905149.005505896]: Failed to read state from robot. Shutting down!
================================================================================REQUIRED process [kuka_hardware_interface-1] has died!
process has finished cleanly
log file: /home/maxgitt/.ros/log/1754bf02-94d7-11e7-aba9-509a4c18ad9f/kuka_hardware_interface-1*.log
Initiating shutdown!
================================================================================
```

Sending a JointTrajectoryGoal works in simulation, so I know it's a realtime issue. But if RSI isn't able to read commands quick enough at the highest priority level than I'm unsure what to do. Should I be giving another process priority as well?

maxgittelman, 2017-09-14:
There is 1 issue that I would like to bring up that may be causing me issues. Whether I run in simulation or on the hardware, rviz shows me that the robot appears in a very strange position. You can by looking below at the image and the position/orientation of each link:

[image]

It is only once I launch `rqt_joint_trajectory_controller`, even before changing any values on the slider, the links naturally move to a normal position and orientation. You can visualize it and see how the values change below:

[image]

1. **I believe the hardware shuts me out because it doesn't accept the starting pose**
2. If so, where do I set the starting positions:
    - I have looked at `ros_rsi.rsi.xml`
    - I have looked at the `xacro` file

Any help would be appreciated, I have been struggling at my job to get this right for nearly 2 weeks with no progress.

SebNag, 2017-09-15:
AFAIK runnig `ros_rsi.src` on the Kuka manipulator sets the arm in a [predefined position](https://github.com/ros-industrial/kuka_experimental/blob/47cd85978a6b7a3d5f227f0cab3fe91d93b84c4a/kuka_rsi_hw_interface/krl/ros_rsi.src#L69) . After the RSI is started on the Kuka Arm the node launched on the ROS computer automatically reads in the current joint values and updates the joint values on the computer according to the real robot. So this behavior should be normal. 
However, I had a problem that the joints of the arm internally weren't correctly calibrated. Thus, when I wanted the robot to move to a position, e.g. Link3 90°, the real robot moved to 90° but the final position of the arm and the position displayed in RViz did not fit. A guy from Kuka had to re-calibrate the robot and "teach" it in what position the arm should be when e.g. Link3 is 90°. 

However, I did not worked with the kuka packages for a few weeks ( and not quite in depth ). 
Maybe @gavanderhoorn has an idea / can provide some help?

maxgittelman, 2017-09-15:
Solved. Using RT Kernel was not necessary. In "T1" mode, by attempting to move to positions that are very far from the start position, the JointTrajectory path would require the robot to accelerate faster than allowed in T1 mode. This would cause the kill switch.

maxgittelman, 2017-09-15:
I might add that their was a discrepancy in the min/max values between the ros_rsi.rsi.xml values and in my xacro (URDF) file. Maybe this played a smaller part.

gavanderhoorn, 2017-09-26:
@maxgittelman: your rviz screenshots make me think that your urdf isn't created correctly. KUKAs in their 'zero pose' are completely stretched out (ie: pointing forward, along +x), they are not in the typical 'hooked' pose that most other robots have (with elbow at 90).

See the `KR 210` model we have in this repository for an example.

Perhaps that is something to check.

maxgittelman, 2017-09-26:
@gavanderhoorn, yes unfortunately I cannot figure out how to correct my URDF of the `KR 500` that I work with. I have considered using the `KR 210`, but am aware that they have different configurations, and am unsure if this would work.

I have even downloaded the STEP file of the KR500 from Kuka's website and converted to URDF via the `ROS solidworks URDF exporter plugin`. However, the website only gives me 1 link to represent the entire robot instead of the correct 6 links. 

Do you know where I can retrieve a properly configured URDF for the KR500?

gavanderhoorn, 2017-09-26:
> Do you know where I can retrieve a properly configured URDF for the KR500?

I don't know of any existing KR500 support packages I'm afraid. If you can tell me which particular variant you have I can see whether I can find something.

> I have considered using the `KR 210`, but am aware that they have different configurations, and am unsure if this would work.

I only mentioned the 210 as an example of what the robot pose would be when you put it in an all-zeros joint space pose.

maxgittelman, 2017-09-26:
I have the `KR500_3 C4 FLR ZH500_3`

gavanderhoorn, 2017-09-27:
I've pushed a support package for the KR 500-3 to my personal fork. See [gavanderhoorn/kuka_experimental/kuka_kr500_support](https://github.com/gavanderhoorn/kuka_experimental/tree/kuka_kr500_support).

Let me know if that works for you. That model has the zero pose as it should be.

Pay extra attention to joint rotation directions as I cannot test those here.

Also: please check whether this model renders correctly and is not too heavy on RViz. It's got some very smooth meshes, but they may be a bit too much (I'd rather have RViz do useful things instead of rendering my 100k+ faces robot model).

gavanderhoorn, 2017-09-27:
And could I ask you to open a new issue if you run into anything else? Thanks.

maxgittelman, 2017-09-27:
Thank you so much. And of course, I'll open a new issue.

gavanderhoorn, 2017-10-01:
@maxgittelman: have you had a chance to test the support pkg?

maxgittelman, 2017-10-02:
yes, it works perfectly! Thank you so much for pushing the package

emielke12, 2022-02-24:
> I've pushed a support package for the KR 500-3 to my personal fork. See [gavanderhoorn/kuka_experimental/kuka_kr500_support](https://github.com/gavanderhoorn/kuka_experimental/tree/kuka_kr500_support).

@gavanderhoorn can I ask how you made this model? I need the kr500 r2830 MT model, and would rather not reinvent the wheel here.