ros-industrial/kuka_experimental #85: How to add a sensor?

AAvi, 2017-05-20:
Hey,

How should one add a sensor? Should it be connected through EtherCAT directly to the controller and then sent back to the computer in the string from the controller, or through a separated UDP connection, or just USB?
What should facilitate real time correction using the sensor? 

Thank you

closed 2017-06-09 by AARobotics

gavanderhoorn, 2017-06-10:
My apologies for my delayed response, this had fallen off my radar somehow.

> How should one add a sensor? Should it be connected through EtherCAT directly to the controller and then sent back to the computer in the string from the controller, or through a separated UDP connection, or just USB?
> What should facilitate real time correction using the sensor?

As you probably found out, you can edit the RSI configuration that is used by `kuka_rsi_hw_interface` just as you could any other regular RSI config. Whatever you add there will become part of the XML broadcast by your controller.

We have no built-in support for any of this, but a small modification the the right nodes would allow you to expose this additional info to your ROS node graph.

note: if you run a real-time kernel and wish to maintain deterministic behaviour don't use regular publishers, but make sure to use the `realtime_tools` provided one.

gavanderhoorn, 2017-06-10:
Minor: could I please ask you to not edit your OP but instead just reply to your own issue? That way we keep a coherent issue tracker, which is valuable for future readers.

AARobotics, 2017-06-13:
Thanks gavanderhoorn!
Are you available for outsourcing? we can use your assistance.  Couldnt find your email.