ros-industrial/kuka_experimental #86: support: subsample visual meshes

gavanderhoorn, 2017-06-12:
Repository is currently 90+MB, mostly due to large Collada meshes.

Visual detail is nice, but the current meshes are a bit *too* detailed.

Use Blender (*decimate* modifier) to reduce face count a bit, while maintaining mesh 'fidelity'.

BrettHemes, 2017-11-10:
Do you have a target size in mind for these?

BrettHemes, 2017-11-11:
Made some progress on this.  So far have done kr5, kr6, kr16_2, kr120, and lbr_iiwa (see branch [here](https://github.com/BrettHemes/kuka_experimental/commits/model-decimation)).  I decimated all of the visual dae files to the point where there were obvious changes in the "fidelity" and then backing off a bit.  In general the sweet spot was right around a factor of 0.5 using Blender's collapse decimation.  I checked all of the changes via the test launch files and the visual differences are minimal or unnoticeable (in my opinion).

These changes have saved ~45 MB so far (down to 37.4 from 82.3).  The two packages with stuff left to address are the kr210 and lbr_iiwa.  Some comments concerning each:

kr210:
\* The models are strange in this one in that they have many internal occluding meshes...  might need to be remodeled?  This package is 14 MB so maybe another 7 MB to save possibly.

lbr_iiwa:
\* There are lots of big files that seem unneeded in both the `blender` and `original` folders (9.4 and 8.5 MB respectively).  Any reason these are here?  I have removed these in separate commits in case they should be kept.

Finally, one more comment: the undecimated originals will still be in the repo history (including .git folder, the new size is 70 MB versus the original 108 MB) unless they are removed from the history as well...

gavanderhoorn, 2018-05-15:
I believe we can close this now that #102 was merged.

closed 2018-05-15 by gavanderhoorn