ros-industrial/kuka_experimental #88: Use .urdf files in Gazebo

SNBasti, 2017-06-16:
Is there a straightforward approach to port the urdf files ( specifically [KR6 R900 sixx urdf](https://github.com/ros-industrial/kuka_experimental/blob/indigo-devel/kuka_kr6_support/urdf/kr6r900sixx.xacro) ) into Gazebo for simulating the movements of the manipulators? 

I noticed[ this ](http://gazebosim.org/tutorials/?tut=ros_urdf) however on the first view this does not work, since I am missing the `inertia` elements. 

Is there an interest to port the `urdf`-file into `sdf`-files for Gazebo compatibility?

gavanderhoorn, 2017-06-16:
While it may provide for a more integrated experience in some cases, there's actually no need to 'port' the `.xacro`s to sdf to make them usable in Gazebo.

The `gazebo_ros_pkgs` support ROS-Gazebo integration and provide a 'spawner' type node/binary that will allow you to instantiate models inside a running Gazebo instance based on a urdf parameter.

What is needed indeed is that the urdf model needs to have at least `inertial` children for `link`s and `dynamics` children for `joint`s. Those will have to be added, but urdf supports that. After that, depending on the amount of ROS integration desired, `gazebo_ros_control` can be added. That would require `transmission`s and some other things.

Personally I like to keep the models in `_support` packages as clean as possible, so keeping Gazebo-specific things as much as possible out of the base models would be preferred (if only so users that prefer V-REP fi do not have to start editing base model files to get rid of Gazebo specifics).

It's not too difficult to do this, and for some models over at `ros-industrial/abb_experimental` and `ros-industrial/staubli_experimental` we've already done that. See [abb_experimental/abb_irb120_gazebo](https://github.com/ros-industrial/abb_experimental/tree/kinetic-devel/abb_irb120_gazebo) and [staubli_experimental/staubli_tx90_gazebo](https://github.com/ros-industrial/staubli_experimental/tree/indigo-devel/staubli_tx90_gazebo) for example.

The idea with these packages is to put all Gazebo-specifics in the the `_gazebo` pkgs, and only add the minimal (and fairly simulator-independent) inertia and dynamics properties to the `_support` pkgs (as those last two cannot be added using Gazebo's `link` reference system).

gavanderhoorn, 2017-06-21:
Can this be closed now @SNBasti?

closed 2017-06-21 by SNBasti

SNBasti, 2017-08-03:
I created a [working experimental setup](https://github.com/SNBasti/kuka_kr6r900sixx_moveit_gazebo_setup) for the kuka kr6r900sixx ( moveit and gazebo ) based on [your moveit configuration branch](https://github.com/gavanderhoorn/kuka_experimental/tree/kr6r900sixx_moveit_rsi_convenience) maybe this is of any interest for someone looking up this issue.

gavanderhoorn, 2017-08-03:
Very nice. Thanks for letting us know.