ros-industrial/kuka_experimental #90: rsi_hw_iface: add faq or troubleshooting guide

gavanderhoorn, 2017-06-21:
There have been a few recurring questions about RSI and the hw interface that might make sense to be put in a FAQ like document. The `krl/readme` already has a small section / some content, that should probably be separated out into a dedicated file / page.

This would make it easier for users to find this information.

gavanderhoorn, 2017-06-21:
Related: #70, #71, #75 and #89.

BrettHemes, 2017-11-11:
I started a page in the usual place: http://wiki.ros.org/kuka_rsi_hw_interface/Troubleshooting
Comments/changes welcome.  I started with #70, #71, and #75

gavanderhoorn, 2017-11-16:
Thanks @BrettHemes. The `Troubleshooting` page is a nice place to put this sort of information. We just need to make sure we include a link to it in the relevant places.

Closing this one.

Additional suggestions for items on the `Troubleshooting` page can be done in new issues.

closed 2017-11-16 by gavanderhoorn