ros-industrial/kuka_experimental #93: rsi_hw_interface.krl

Jie-X, 2017-08-03:
There is a kuka robot kr6r700sixx with KR_C4. I want to connect kuka to ros, In the configuration of controller network : 

> 4. Run RSI-Network from the Windows Start menu (All Programs > RSI-Network).

But there is no RSI-Network in all programs, to be seriously, I can't find any introduction about RSI in the robot user manual. Can you give me some help?

cschindlbeck, 2017-08-03:
The Robot-Sensor-Interface (RSI) is software from KUKA that has to be bought seperately. Check if you bought that software.

Jie-X, 2017-08-03:
Maybe I haven't bought it. There is a network configuration in HMI, can it be used for communication with ros?

gavanderhoorn, 2017-08-03:
We don't currently have any other interfacing component other than the RSI `hardware_interface` in this repository.

See #77 and #80 for related questions.

gavanderhoorn, 2017-08-08:
I'm closing this, as it's not an issue with the packages in this repository.

If you feel something still needs to be discussed @Jie-X, please reopen the issue.

closed 2017-08-08 by gavanderhoorn

Jie-X, 2017-08-09:
sorry, I have encountered some new problems. 

> 9. Run **cmd.exe** and ping the PC you want to communicate with on the same subnet

It's ok for PC to ping RSI, however RSI can not ping PC from the controller. Will this impact communication? And his thing also happen to KLI(KUKA Line Interface), despite it, I can download program to controller by WorkVisual.

KLI: 
- [IP] 192.168.1.17 
- [mask] 255.255.255.0

RSI:
- [IP] 192.168.2.10 
- [mask] 255.255.255.0 

PC:
- [IP] 192.168.2.11
- [mask] 255.255.255.0

gavanderhoorn, 2017-08-09:
It's unlikely that you can ping in one direction, but not in the other. `ping` requires the IP stack to work in both ways.

It is possible for PCs to ignore ICMP messages though. Can you check that you don't have a firewall on your PC that is dropping all the traffic?

gavanderhoorn, 2017-08-09:
Btw: this is not really an issue with these packages, I would recommend that you try to find some more specialised help to get your network configuration to work.

Jie-X, 2017-08-09:
Well. Thx.