ros-industrial/kuka_experimental #94: questions on command torque, and support

Thompson104, 2017-08-08:
Hey Guys,

I'm a graduate student at mechanical engineering. We brought a UR5 last year but it happens that it doesn't give us access to its joint torque. We are going to implement some hybrid force/motion controller and we are not sure if we should get kuka or fanuc. I wonder if you guys could help us answer the following question. I know some similar questions have been answered before but it has been a long time and I want to get an accurate answer, since these robots are not cheap:( . Thank you in advance.

1. Do we have direct control over the joint torque for KUKA KR16? Do we need to calculate and compensate for the gravity by ourselves?

2. Is the driver for KUKA supported by KUKA or is this driver developed by some researchers? I asked this question because UR_ros_driver sends URScript(programming language for UR) under the hood, so basically ros_industry doesn't offer much to UR driver. We want to utilize ros_control and possible ROS 2 in the future. 

3. Which robot vendor would you recommend KUKA or FANUC for force control? We want to get robot with high payload(more than 13kg). It seems FANUC offers official ros support. I'm not sure which one is better.

I know I may ask some trivial questions. But it is kind of important to us since these robots are really expensive. I'll do my research but your review is really important to us. Please help me to clarify this questions. I'd really really appreciate your help.

gavanderhoorn, 2017-08-08:
Some things to set straight here.

Also: we're happy to answer your questions, but these questions would really have been better asked on the ROS-Industrial mailing list, not an issue on the KUKA experimental repository.

> Do we have direct control over the joint torque for KUKA KR16? Do we need to calculate and compensate for the gravity by ourselves?

KR C4 with RSI only supports position control, so no velocity nor torque. Which specific manipulator you pair with the controller does not influence this.

This is a vendor limitation, not a limitation of the ROS components.

> Is the driver for KUKA supported by KUKA or is this driver developed by some researchers?

No KUKA involvement - unfortunately we're "just some researchers" here.

> I asked this question because UR_ros_driver sends URScript(programming language for UR) under the hood, [..]

I don't really understand what you're trying to say here. The only way to interface with UR controllers is through sending URScript. All external motion interfaces to UR controllers will use that, not just `ur_driver` or `ur_modern_driver`.

The only other option is LLSI, but that is going to be involved and is far from user friendly. You're free to implement something using that, if it serves your needs. For most of our use-cases -- and those of the majority of users -- the convenience of using URScript and being able to use the TP normally apparently outweighs the benefit of using an LLSI based driver.

> so basically ros_industry doesn't offer much to UR driver. 

I'm not sure what this means.

> Which robot vendor would you recommend KUKA or FANUC for force control? We want to get robot with high payload(more than 13kg). It seems FANUC offers official ros support. I'm not sure which one is better.

This will really depend on what you *actually* want to do:

 1. joint space level torque control over all joints
 1. 'fast enough' position control to implement a poor-mans force control using input from a f/t sensor

Very few industrial robots support joint space torque control - especially if you want to use that on the "factory floor". For research, there is KUKA FRI (older LBRs - EoL) and the IIWAs support compliance / impedance control (but that's not the same).

AFAIK none of the big-four mfgs allow you access to their motion controllers at the joint torque level.

FANUC does not have any usable external motion interfaces (well, perhaps DPM). ABB does not support joint torque control. Neither does Yaskawa.

If you can be more specific I can perhaps suggest something or nuance my answer a bit.

gavanderhoorn, 2017-08-08:
And I just noticed this:

>  It seems FANUC offers official ros support

Can you please tell me what made you believe that Fanuc has official ROS support?

Fanuc's OS is called *ROS*, but that is the *Realtime OS*, not the *Robot Operating System*. Don't confuse the two.

`fanuc_driver` and all the packages under `ros-industrial/fanuc` and `ros-industrial/fanuc_experimental` are again created by "just some researchers" I'm afraid.

Thompson104, 2017-08-08:
@gavanderhoorn  Thank you so much for your reply.

But please don't get me wrong. I didn't mean any disrespect for you guys. I'm glad you can help. But I don't think your sarcasm is funny here.  and I said 

> is this driver developed by some researchers? 

not 

> just some researchers

I said 

>  It seems FANUC offers official ros support.

I guess "It seems" means I'm not sure. I called them and they said they may able to help us with our ROS projects. You may want to confirm this with them. 

What I mean by 

> so basically ros_industry doesn't offer much to UR driver.

is that ros_industry couldn't help us much with force control methods. My English may not be precise and  I added (force control) to make it more clear. I apologize for not making this part clear. 

Basically we are trying to replicated [godel](https://github.com/ros-industrial-consortium/godel) but we also want to maintain a constant contact force on the polishing surface. We are looking for industry robots with payloads more than 13kg that may help us with this. Again I'm sorry if what I said offend you. In my opinion, I'm not sure why "some researchers" may cause such confusion. I hold high regard for researchers and I sincerely don't get your point by "just some researchers"

gavanderhoorn, 2017-08-09:
@Thompson104 wrote:
> @gavanderhoorn  Thank you so much for your reply.
> 
> But please don't get me wrong. I didn't mean any disrespect for you guys. I'm glad you can help. But I don't think your sarcasm is funny here.  and I said 
> 
>> is this driver developed by some researchers? 
> 
> not 
> 
>> just some researchers

This may have been a combination of non-native speakers and my general understanding of your question(s). My apologies if my reply came of harsher than it should have.

> I said 
> 
>>  It seems FANUC offers official ros support.
> 
> I guess "It seems" means I'm not sure. I called them and they said they may able to help us with our ROS projects. You may want to confirm this with them. 

And that is why I asked if you could tell me what made you think that.

Especially since there are large regional differences between Fanuc offices in different countries, and we've had situations before where mfgs suddenly claim to support all sorts of things while in reality they don't.

> What I mean by 
> 
> > so basically ros_industry doesn't offer much to UR driver.
> 
> is that ros_industry couldn't help us much with force control methods. My English may not be precise and  I added (force control) to make it more clear. I apologize for not making this part clear. 

Joint level torque control with industrial robots is just something that is not well supported, anywhere, no matter the interface you use. That is not something we can change.

Or at least not joint level torque control.

> Basically we are trying to replicated [godel](https://github.com/ros-industrial-consortium/godel) but we also want to maintain a constant contact force on the polishing surface. We are looking for industry robots with payloads more than 13kg that may help us with this. 

Cartesian compliance / force control is a different thing all together, and is something that is supported by the majority of industrial robots. You don't need joint torque control for that. This is also why I asked what you actually wanted to do.

We do not have support for that in our current drivers, so that will take some work, but it should be possible.

> Again I'm sorry if what I said offend you. In my opinion, I'm not sure why "some researchers" may cause such confusion. I hold high regard for researchers and I sincerely don't get your point by "just some researchers"

"some researchers", especially when you put "is this supported by the manufacturer" before it can be understood to carry a derogatory or sarcastic sentiment. It doesn't just mean "no researchers in particular" (ie: they can come from anywhere), which I now believe you tried to say.

gavanderhoorn, 2017-09-26:
I'm closing this as I believe we've addressed the questions of the OP.

@Thompson104: if you feel this is incorrect, please reopen.

closed 2017-09-26 by gavanderhoorn