ros-industrial/kuka_experimental #96: rsi_hw_interface: KSS1073 KSS1074

Jie-X, 2017-08-29:
Context:
- robot: KR6R700sixx
- controller: KR C4 (v8.3.26)
- RSI: 3.2
- mode: T1 or AUTO

I have met the same trouble as #89. After enabling the rqt controller (push the big red button), robot trembles slightly at home point(`0, -90, 90, 0, 90, 0`) and gives me `KSS1073: Command gear torque An` error later. As long as I move the slider a little bit, the robot will give me `KSS1073 and KSS1074` error and stop immediately. Did you solve this problem? @gavanderhoorn

Jie-X, 2017-09-01:
Hey, this has been troubling me for a long time. Can you give me some advice?
The Ubuntu with ROS is installed on virtual machine. A series of DUPLICATE will be returned if I ping RSI from Ubuntu. Will this disturb the communication?
```
PING 172.30.1.110 (172.30.1.110) 56(84) bytes of data.
64 bytes from 172.30.1.110: icmp_seq=1 ttl=255 time=0.497 ms
64 bytes from 172.30.1.110: icmp_seq=1 ttl=64 time=0.507 ms (DUP!)
64 bytes from 172.30.1.110: icmp_seq=2 ttl=255 time=0.589 ms
64 bytes from 172.30.1.110: icmp_seq=2 ttl=64 time=0.614 ms (DUP!)
64 bytes from 172.30.1.110: icmp_seq=3 ttl=255 time=0.906 ms
64 bytes from 172.30.1.110: icmp_seq=3 ttl=64 time=0.933 ms (DUP!)
64 bytes from 172.30.1.110: icmp_seq=4 ttl=255 time=0.546 ms
64 bytes from 172.30.1.110: icmp_seq=4 ttl=64 time=0.565 ms (DUP!)
64 bytes from 172.30.1.110: icmp_seq=5 ttl=255 time=0.642 ms
64 bytes from 172.30.1.110: icmp_seq=5 ttl=64 time=0.669 ms (DUP!)
--- 172.30.1.110 ping statistics ---
5 packets transmitted, 5 received, +5 duplicates, 0% packet loss, time 4000ms
rtt min/avg/max/mdev = 0.497/0.646/0.933/0.149 ms
```
Please reply to me if you are free. Thank you.

gavanderhoorn, 2017-09-01:
It's up to you, but I would never run anything communicating with my KR C4 over RSI in a Virtual Machine. RSI is rather sensitive to proper timing on the PC end, and seeing the problems you mention in your OP, I would first try to migrate to a native / dual boot setup.

gavanderhoorn, 2017-09-01:
And I'm really sorry, but I'm afraid I can't help you with getting OS and / or networking issues resolved, as I'm pretty stretched already - dealing with just ROS and ROS-Industrial issues.

I'd recommend you try other fora which deal with those kind of things more specifically.

Jie-X, 2017-09-04:
DUPLICATE will still be returned in the real operating system, but the interface works well. Thank you very much.

closed 2017-09-04 by Jie-X