ros-industrial/kuka_experimental #97: How can I get the real D-H parameter？

csdjywyq, 2017-11-29:
I want to use the kr210 package make a moveit! to plan the trajectory. But what I have found is that the D-H parameter in the source files is not same with the real robot that I'm using because I found that the Cartesian position showed on the robot pad is similar to but not the same as this showed in tf. So I want to ask how can I get the real D-H parameter of the real robot?
thank you !

gavanderhoorn, 2017-11-29:
> I want to use the kr210 package make a moveit! to plan the trajectory. But what I have found is that the D-H parameter in the source files is not same with the real robot that I'm using because I [...]

The urdf of the KR210 is not based on the D-H parameters, so I'm not surprised the poses you see on the TP deviate from those that ROS gives you.

That is a problem though and we should rectify it. 

As you can see in the issue tracker we have a nr of issues with the KR210 pkg, so it can defenitely benefit from some work.

> So I want to ask how can I get the real D-H parameter of the real robot?

Where to get the DH parameters is a bit of an orthogonal problem and not really something for this tracker.

You could see whether you can find them in a paper or perhaps someone else has made them available in some for or another. Unfortunately we don't have them here. And I seem to remember that KUKA also doesn't use them, so that may be why they are hard to come by (ie: because someone has to do the work).

gavanderhoorn, 2017-11-29:
I'm going to close this issue (as we don't use D-H parameters and the xacro doesn't contain them), but I did capture the fact that the kinematic chain in the current xacro for the KR210 does not correspond to the actual model of the robot.

closed 2017-11-29 by gavanderhoorn