ros-industrial/kuka_experimental #101: Controller compatibility and roslaunch error

Dmitryvoit, 2018-01-22:
1) Is there any way to control Kuka KR 60 HA with ROS at the current moment? And also,since KR 120 in supported robots, is it possible to work with KR 120 HA R2700?

2) I tried to start test_hardware_interface.launch sim:=false, and got his: 
[image]

and then 

[image]

Any Ideas what happened?

gavanderhoorn, 2018-01-22:
Have you followed the readme for either the [KR C2](https://github.com/ros-industrial/kuka_experimental/tree/indigo-devel/kuka_rsi_hw_interface/krl/KR_C2) or [KR C4](https://github.com/ros-industrial/kuka_experimental/tree/indigo-devel/kuka_rsi_hw_interface/krl/KR_C4)?

gavanderhoorn, 2018-01-22:
Could I also please ask you to not post screenshots (ie: *images*) of console text? It's really just text, so please copy-paste it into the issue directly, and add ``` above and below.

gavanderhoorn, 2018-01-22:
@Dmitryvoit wrote:
> At the step 3, I didn't understand, how to do this:

> > In order to successfully launch the kuka_rsi_hardware_interface a parameter robot_description needs to be present on the ROS parameter server. This parameter can be set manually or by adding this line inside the launch file (replace support package and .xacro to match your application):

> Do I need to do it even for test?

yes.

The component uses that information to figure out joints and limits.

Dmitryvoit, 2018-01-22:
Well, just added 

> <param name="robot_description" command="$(find xacro)/xacro.py '$(find kuka_kr6_support)/urdf/kr6r900sixx.xacro'"/>

And after I launched 

> roslaunch kuka_rsi_hw_interface test_hardware_interface.launch sim:=false

I got his:

> madrobot@madrobot:~$ roslaunch kuka_rsi_hw_interface test_hardware_interface.launch sim:=false
... logging to /home/madrobot/.ros/log/db0a34b8-ff7b-11e7-97d0-0800270771f6/roslaunch-madrobot-27668.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

Traceback (most recent call last):
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/__init__.py", line 306, in main
    p.start()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 268, in start
    self._start_infrastructure()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 217, in _start_infrastructure
    self._load_config()
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/parent.py", line 132, in _load_config
    roslaunch_strs=self.roslaunch_strs, verbose=self.verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/config.py", line 451, in load_config_default
    loader.load(f, config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 748, in load
    self._load_launch(launch, ros_config, is_core=core, filename=filename, argv=argv, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 720, in _load_launch
    self._recurse_load(ros_config, launch.childNodes, self.root_context, None, is_core, verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 664, in _recurse_load
    self._param_tag(tag, context, ros_config, verbose=verbose)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 95, in call
    return f(*args, **kwds)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 256, in _param_tag
    vals = self.opt_attrs(tag, context, ('value', 'textfile', 'binfile', 'command'))
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 202, in opt_attrs
    return [self.resolve_args(tag_value(tag,a), context) for a in attrs]
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/xmlloader.py", line 183, in resolve_args
    return substitution_args.resolve_args(args, context=context.resolve_dict, resolve_anon=self.resolve_anon)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 370, in resolve_args
    resolved = _resolve_args(resolved, context, resolve_anon, commands)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 383, in _resolve_args
    resolved = commands[command](resolved, a, args, context)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 151, in _find
    source_path_to_packages=source_path_to_packages)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/roslaunch/substitution_args.py", line 197, in _find_executable
    full_path = _get_executable_path(rp.get_path(args[0]), path)
  File "/usr/lib/python2.7/dist-packages/rospkg/rospack.py", line 203, in get_path
    raise ResourceNotFound(name, ros_paths=self._ros_paths)
ResourceNotFound: kuka_kr6_support
ROS path [0]=/opt/ros/kinetic/share/ros
ROS path [1]=/home/madrobot/catkin_ws/src/fake_ar_publisher
ROS path [2]=/home/madrobot/catkin_ws/src/kuka_rsi_hw_interface
ROS path [3]=/home/madrobot/catkin_ws/src/myworkcell_core
ROS path [4]=/opt/ros/kinetic/share

Is that even ok? No errors,but nothing happens also

gavanderhoorn, 2018-01-22:
From this:

> I compiled kuka_rsi_hw_interface package

and the error you post I get the impression that you did not build all the packages in `kuka_experimental`.

Please just build everything using `catkin_make` or `catkin build` in your workspace, `source` the appropriate `setup.bash` and then try again.

Dmitryvoit, 2018-01-23:
Well, there is my launch file:

> <?xml version="1.0"?>
<launch>
  
 <param name="robot_description" command="$(find xacro)/xacro.py '$(find kuka_rsi_hw_interface)/urdf/kr120r2500pro.xacro'"/>
</launch>

And there is the output after I launched it:

> <?xml version="1.0"?>
<launch>
  
 <param name="robot_description" command="$(find xacro)/xacro.py '$(find kuka_rsi_hw_interface)/urdf/kr120r2500pro.xacro'"/>
</launch>

gavanderhoorn, 2018-01-23:
As I already wrote in https://github.com/ros-industrial/kuka_experimental/issues/101#issuecomment-359398381, please format console copy-pastes, program code, xml and similar text in your question using the *Insert code* button (it's the 2nd one in the middle of the second triple of the formatting bar: `< >`.

gavanderhoorn, 2018-01-23:
The error message you include says: `resource not found: kuka_kr120_support`.

So `xacro` is looking for a package with that name.

From your statement:

> inside `kuka_rsi_hw_interface` there is a folder URDF so it should be visible

And:

```
when processing file: /home/madrobot/catkin_ws/src/kuka_rsi_hw_interface/urdf/kr120r2500pro.xacro
```

I have the impression that you copied `kr120r2500pro.xacro` from `kuka_kr120_support/urdf` to `kuka_rsi_hw_interface/urdf`.

Without updating the xacro (and the macro) to not refer to `kuka_kr120_support` anymore, that won't work.

Is there any reason you cannot just build all packages in this repository? That should make everything start working immediately.

Dmitryvoit, 2018-01-23:
I tried to, but got these errors:
[CMakeError.log](https://github.com/ros-industrial/kuka_experimental/files/1655357/CMakeError.log)
[CMakeOutput.log](https://github.com/ros-industrial/kuka_experimental/files/1655358/CMakeOutput.log)

gavanderhoorn, 2018-01-23:
I cannot quickly see any errors in the two files you attach.

Dmitryvoit, 2018-01-23:
Got it. I started the KR-120 simulation with this:

> roslaunch kuka_kr120_support test_kr120r2500pro.launch sim:=false

So, in  kuka_kr120_support/config there is only one file (joint_names). What should I do now, copy hardware_controllers from kuka_rsi_hw_controller to the kuka_kr120_support/config folder, Or I can just change .yaml files in kuka_rsi_hw_controller/config folder and then run the kuka_rsi_hw_controller?

gavanderhoorn, 2018-01-23:
You don't need to copy anything - yet.

To test things, what you need is a `controller_joint_names` parameter and the `ros_control` parameters that you can find in the `kuka_rsi_hw_interface/test` folder.

Setting this up for a new robot is a bit underdocumented at the moment. It helps if you already have experience with `ros_control` and ROS in general.

Documenting this better is going to depend on a community contribution, I won't have time for it soon.

Dmitryvoit, 2018-01-23:
> Setting this up for a new robot is a bit underdocumented at the moment. It helps if you already have experience with ros_control and ROS in general.

You mean, If I want to test it with robots that currently are not supporting?

gavanderhoorn, 2018-01-23:
Yes.

But do understand: RSI (and its configuration) (almost) doesn't care about the specific robot you have connected to your controller, and neither does the hw interface node. All it needs are joint names and some `ros_control` parameters.

re: no support: 'support' in these cases is a very loosely defined concept: the driver itself is almost manipulator-agnostic, so to support a new variant or series is almost trivial. If you want a nice visualisation and MoveIt compatibility you'll have to create a urdf and meshes etc.

Dmitryvoit, 2018-01-23:
I see,thank you very much for explaining and your patience!
But If,for example, I have KR-120, and would like to establish connection to it  - where should I change the IP address of the robot? Do I need to pass all the steps,that are mentioned there:

> https://github.com/ros-industrial/kuka_experimental/tree/indigo-devel/kuka_rsi_hw_interface/krl/KR_C4

gavanderhoorn, 2018-01-23:
I would recommend you follow the readme, at least once, yes.

Especially if you are not too experienced with ROS/`ros_control`/RSI etc it's probably a good basis.

Dmitryvoit, 2018-01-23:
But I can't get it, how If i will launch kuka_kr120_support/launch/load_kr120r2500pro.launch it will get the IP adress? from which files? From the .yaml's in  kuka_rsi_hw_controller/config?

gavanderhoorn, 2018-01-23:
This is mentioned in section 3 of the KR C4 readme.

See [kuka_rsi_hw_interface/test/test_params.yaml](https://github.com/ros-industrial/kuka_experimental/blob/8f05d95542880ea640a3e7cc1183066e109cf6bf/kuka_rsi_hw_interface/test/test_params.yaml), which is loaded by [kuka_rsi_hw_interface/test/test_hardware_interface.launch](https://github.com/ros-industrial/kuka_experimental/blob/8f05d95542880ea640a3e7cc1183066e109cf6bf/kuka_rsi_hw_interface/test/test_hardware_interface.launch#L6).

Dmitryvoit, 2018-01-23:
Yeah, I saw that,but what the point with kuka_kr120_support package then? just compatibility with moveit and other visual packages?

gavanderhoorn, 2018-01-23:
`kuka_kr120_support/launch/load_kr120r2500pro.launch` populates the `robot_description` parameter with the `xacro` of the KR 120 R2500 Pro.

gavanderhoorn, 2018-01-23:
Please see [Working with ROS-Industrial Robot Support Packages](http://wiki.ros.org/Industrial/Tutorials/WorkingWithRosIndustrialRobotSupportPackages) for an overview of how we structure these things in ROS-I in general.

gavanderhoorn, 2018-01-23:
@Dmitryvoit wrote:
> Yeah, I saw that,but what the point with kuka_kr120_support package then? just compatibility with moveit and other visual packages?

I would say that if you use a robot with ROS, but don't have a urdf, you're not doing it right. So the point of those packages is to provide those models and files.

Without those, you can't really do anything useful.

Dmitryvoit, 2018-01-23:
The last question,sir.
After I launched 

> roslaunch kuka_rsi_hw_interface test_hardware_interface.launch sim:=false

as in Readme, in terminal I got this: 

> [INFO] [1516708629.532628]: Loading controller: position_trajectory_controller
> [ERROR] [1516708629.534796918]: Could not load controller 'position_trajectory_controller' because controller type 'position_controllers/JointTrajectoryController' does not exist.
> [ERROR] [1516708629.534819057]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
> [ERROR] [1516708630.536207]: Failed to load position_trajectory_controller
> [INFO] [1516708630.536426]: Loading controller: joint_state_controller
> [ERROR] [1516708630.538379627]: Could not load controller 'joint_state_controller' because controller type 'joint_state_controller/JointStateController' does not exist.
> [ERROR] [1516708630.538435187]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
> [ERROR] [1516708631.539776]: Failed to load joint_state_controller

And now seems like it waiting for robot connection. Are these errors because I didn't connect robot or it's something else? 

 I added

> (<param name="robot_description" command="$(find xacro)/xacro.py '$(find kuka_kr120_support)/urdf/kr120r2500pro.xacro'"/>) (withouht brackets)

to the test_hardware_interface.launch, also I changed IP in test_params.yaml (my laptop's IP address),and in ag1_params.yaml (also added IP address of the laptop)

BrettHemes, 2018-01-24:
> [ERROR] [1516708629.534796918]: Could not load controller 'position_trajectory_controller' because controller type 'position_controllers/JointTrajectoryController' does not exist.

> [ERROR] [1516708630.538379627]: Could not load controller 'joint_state_controller' because controller type 'joint_state_controller/JointStateController' does not exist.

You are missing some of the required controller packages.  Assuming you are in Ubuntu on Kinetic, try installing them via the following:
`$ sudo apt-get install ros-kinetic-joint-trajectory-controller`
`$ sudo apt-get install ros-kinetic-joint-state-controller`

Dmitryvoit, 2018-01-25:
> You are missing some of the required controller packages. Assuming you are in Ubuntu on Kinetic, try installing them via the following:
> $ sudo apt-get install ros-kinetic-joint-trajectory-controller
> $ sudo apt-get install ros-kinetic-joint-state-controller

Thank you kindly,sir,that worked! Going to test on live robot today.

gavanderhoorn, 2018-01-25:
@Dmitryvoit: did you run the `rosdep install --from-paths ..` command before building your workspace as is shown in [#q186679](https://answers.ros.org/question/186679) (ROS Answers) that you linked in https://github.com/ros-industrial/kuka_experimental/issues/101#issuecomment-359738352?

If you did, and those two packages were not installed, it would seem we have some missing run dependencies in one of our packages.

Dmitryvoit, 2018-01-25:
@gavanderhoorn  Well, I didn't remember exactly, because I was working on other virtual machine,that have  crushed hardly, so, I just made an experiment. In fresh ubuntu I installed ROS kinetic, cloned kuka_experimental from git, did all that mentioned there: 
https://answers.ros.org/question/186679/problems-with-install-hector-quadrotor/
Then added into launch file this just for  the test:

> <param name="robot_description" command="$(find xacro)/xacro.py '$(find kuka_kr6_support)/urdf/kr6r900sixx.xacro'"/>

After that tried to start launch file and got the same errors as I had yesterday before I did 

> $ sudo apt-get install ros-kinetic-joint-trajectory-controller
> $ sudo apt-get install ros-kinetic-joint-state-controller

After I ran these commands in new virtual machine everything worked.

gavanderhoorn, 2018-01-25:
I checked the pkg manifests, and we don't explicitly list a dependency on those two controllers anywhere, so that is expected.

The 'problem' is that it's not clear where the dependency should be added: there is absolutely no requirement for anyone using `kuka_rsi_hw_interface` to have either of those two packages (as you can use it without them), so I'm not sure.

I'll open a new issue to track this.

Thanks for confirming.

gavanderhoorn, 2018-01-25:
@Dmitryvoit wrote:
> [..] in new virtual machine everything worked.

Note that previous users have tried using virtual machines with RSI and have ran into various issues.

RSI is very sensitive to dropped packets and performance of the PC and network infrastructure, so be careful.

Dmitryvoit, 2018-01-25:
@gavanderhoorn Thank you very much for all your help! Will try it today on real robot

Dmitryvoit, 2018-01-25:
> Note that previous users have tried using virtual machines with RSI and have ran into various issues.
> 
> RSI is very sensitive to dropped packets and performance of the PC and network infrastructure, so be careful.
Yeah, I read some information about it. Well, If that will not work, I will install linux as second system on my laptop.

gavanderhoorn, 2018-01-25:
I opened #103 to discuss whether we should add dependencies on those two controllers or not.

gavanderhoorn, 2018-03-02:
I have the impression you've been able to get things to work @Dmitryvoit. So I'm closing this.

#103 continues the discussion on the missing dependencies.

If you are still having trouble, feel free to re-open.

closed 2018-03-02 by gavanderhoorn