ros-industrial/kuka_experimental #109: Execution of ros_rsi without

cschindlbeck, 2018-01-31:
Hello,

i want to move the robot by a small amount without first moving into the initial position. Therefore i commented the

` Move to start position
 PTP {A1 0, ...}
`
out and i want to immediately start the RSI_CREATE context

This results in an error 

> KSS01443: Impermissible start motion

How can i realize my goal?

Thanks!

gavanderhoorn, 2018-01-31:
Afaik you need a `PTP` move before starting the RSI context. That is why that line is in there.

gavanderhoorn, 2018-01-31:
You could see if changing it to `PTP` to `$POS_ACT` (or something similar) works.

gavanderhoorn, 2018-01-31:
Note that this is not so much a question about these pkgs btw, so would have been better asked on a KUKA forum.

gavanderhoorn, 2018-01-31:
I'm not sure any more (so verify this), but it could be that the correction (and axes) limits set in the RSI configuration are relative to the starting position. If that is the case, you'll want to update those.

But perhaps another user can confirm or correct this.

BrettHemes, 2018-02-01:
> Afaik you need a move before starting the RSI context

As an aside, some of the terminology is strange in the Kuka universe and additionally some terminology/concepts are more readily found in older versions of the documentation.  But yes, this first move is a "BCO run" (Block COincidence) and serves the purpose of making sure that the robot is where the program expects it to be (i.e., at the block pointer).  This is what is executed on the first press of the green "program start forwards" button.  In general these happen (are needed?) after a program reset, new program selection, program edit, jogging, etc.

> You could see if changing it to PTP to $AXIS_ACT (or something similar) works.

Yup! Despite often being to the `home` position, it can also be to `$axis_act` (current joint targets), `$axis_act_meas` (current joint positions), etc.

> ... but it could be that the correction (and axis) limits set in the RSI configuration are relative to the starting position.

I believe all of the motion objects in the RSI world are thought of as applying corrections to the KRL-guided path.  In our case our KRL "path" is just a single point, therefore ALL of the motion is actually correction from the first `ptp` target ("correction" computed [here](https://github.com/ros-industrial/kuka_experimental/blob/indigo-devel/kuka_rsi_hw_interface/src/kuka_hardware_interface.cpp#L122)).  As an aside it is possible to command the robot from within KRL while the RSI context operates (probably the original assumed use case) so then limits are with respect to the KRL motion setpoints in time.

BrettHemes, 2018-02-01:
> Note that this is not so much a question about these pkgs btw, so would have been better asked on a KUKA forum (I understand why you posted it here though).

@gavanderhoorn Do you have some favorite Kuka forums?

ivareri, 2018-02-01:
IIRC, there is no guarantee that $AXIS_ACT is set until the robot is moved. Might only be after a cold boot, it's some time since I read the manual so I'm a bit hazy on the the details.

Re forum: https://www.robot-forum.com/robotforum/kuka-robot-forum/ is the best I've found.

BrettHemes, 2018-02-01:
I think the act of setting the tool and base frames will update the positional variables.  The 'BAS (#INITMOV,0 )' call in the beginning accomplishes this.

gavanderhoorn, 2018-02-02:
@cschindlbeck: could you let us know what you did in the end?

cschindlbeck, 2018-02-07:
So Kuka support told me that 

`PTP $AXIS_ACT`

will "move" the robot to the current position. However, they told me that after this command the subsequent motion will be executed with full speed. First tests confirm this, however, the same holds if i move the robot in the initial position and then to another position with a PTP motion. So i don't see why it should be a problem. I haven't got the time lately for rigorous testing. Is it true that all motions during the Ethernet connection can be executed with full speed? Or are the velocity limits enforced but not for PTP motions?

BrettHemes, 2018-02-07:
Not exactly sure I understand the question.  Whatever the `$VEL_AXIS` system (global) variable is set to will determine the speed at which `PTP` motions run as a percentage of max joint velocities.  The motion inside of the `RSI_MOVECORR()` is determined by the streamed setpoints in time.  I am not aware of any velocity limits inside of the RSI context besides the actual maximum limits of the system itself.

The default KRL module .src template starts with an inline-form-generated `PTP` to home which sets the `$VEL_AXIS` variable (to 100%) via the call to `BAS(#PTP_PARAMS, 100)`.  The actual unfolded KRL is:

    ;FOLD PTP HOME  Vel= 100 % DEFAULT;%{PE}%MKUKATPBASIS,%CMOVE,%VPTP,%P 1:PTP, 2:HOME, 3:, 5:100, 7:DEFAULT
    $BWDSTART = FALSE
    PDAT_ACT=PDEFAULT
    FDAT_ACT=FHOME
    BAS (#PTP_PARAMS,100 )
    $H_POS=XHOME
    PTP  XHOME
    ;ENDFOLD

Maybe the Kuka rep was referring to this, as it will result in subsequent PTP motions running a full speed?

BrettHemes, 2018-02-07:
> However, they also told me that after this command the subsequent motion will be executed with full speed.

Also, isn't this what you want?

gavanderhoorn, 2018-02-07:
@BrettHemes wrote:
>  I am not aware of any velocity limits inside of the RSI context besides the actual maximum limits of the system itself.

Exactly.

gavanderhoorn, 2018-02-08:
future readers: unfortunately it would seem all of @BretHemes' comments have disappeared from this issue, so it may seem like some info is missing - and that is true: it is.

I have github notification emails of all of it, so I'll see if I'm allowed to reconstruct the conversation.

gavanderhoorn, 2018-03-02:
@cschindlbeck: did moving to the current pose work out for you in the end?

Did you do anything special with the joint limits?

cschindlbeck, 2018-03-02:
I won't have time to test this until mid April so i will close the issue for now and reopen once i have tested it.

closed 2018-03-02 by cschindlbeck

gavanderhoorn, 2018-03-02:
Ah, ok. I thought you'd already made the modifications.