ros-industrial/kuka_experimental #111: Citing ROS-I kuka_experimental in scientific publication?

cschindlbeck, 2018-02-12:
Thanks to everyone involved here, i am in the process of finalizing my first publication with an ROS-I interfaced KUKA robot.

Is there a preferred way of citing this package or ROS-I?

yijiangh, 2018-02-12:
This discussion [1] on ROS-I's google group is definitely related.

[1] https://groups.google.com/forum/#!searchin/swri-ros-pkg-dev/citation%7Csort:date/swri-ros-pkg-dev/a-PXa8CJWb8/0JCB45OuCQAJ

cschindlbeck, 2018-02-13:
Ok, thanks!

closed 2018-02-13 by cschindlbeck

gavanderhoorn, 2018-02-13:
@yijiangh: I cannot find the `edwards2012ros` reference in the article you link. I would say that is the first academic publication about ROS-I.

For ROS itself `M. Quigley, B. Gerkey, K. Conley, J. Faust, T. Foote, J. Leibs, E. Berger, R. Wheeler, A. Ng, "ROS: An open-source robot operating system", Proc. Open-Source Software Workshop Int. Conf. Robotics and Automation, 2009` seems to be the seminal paper.

JeroenDM, 2018-02-13:
@gavanderhoorn Thank you for the ROS reference, I will use that one in the future instead of the website, as I did in the article that @yijiangh mentioned.
(Although the only thing left on the IEEE website seems to be the name of the reference.)

Concerning the mysterious reference `edwards2012ros`, also used in this [1] article, it seems to be untraceable. I think I will use ROS-Industrial's website in the future. Or is this not the right way to do this?

For ROS-I packages I suppose it's ok to use the name, author(s) and wiki page link as a reference.

[1] [https://www.sciencedirect.com/science/article/pii/S2405896317331464](https://www.sciencedirect.com/science/article/pii/S2405896317331464)

gavanderhoorn, 2018-02-13:
@JeroenDM: searching for the ECHORDS 2012 workshop leads to [echord.info/wikis/website/icra-2012-workshop](http://echord.info/wikis/website/icra-2012-workshop), which does not mention @shaun-edwards, but, the [conference proceedings](http://echord.info/file/Documents/WP4-monitoring/T4-4-structured-dialogue/Workshops/ICRA2012/proceedings.pdf) contain the full article (search for `edwards`, it's on page 29).

I'm not certain that is the first academic publication about ROS-I btw.

gavanderhoorn, 2018-02-13:
Referencing the website would probably be better. ROS-I is not really an academic project, so shoe-horning in that paper written by @shaun-edwards seems strange.

The website is where the most up-to-date information about ROS-I can be found, so it's probably better to point readers to that instead of an article from 2012.

JeroenDM, 2018-02-13:
@gavanderhoorn  I did not google long enough, thank you for the link :)