ros-industrial/kuka_experimental #115: problem with KRC

monykhosrvi, 2018-03-05:
Hello everyone, 

currently I am trying to establish an RSI interface between KR6 R900 controller  and the following machine: 

`Linux stamPC 4.4.0-31-generic #50~14.04.1-Ubuntu SMP Wed Jul 13 01:07:32 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux`

Following the below instructions: 
[Configuring RSI on the controller](https://github.com/ros-industrial/kuka_experimental/tree/indigo-devel/kuka_rsi_hw_interface/krl/KR_C4)

I have set the IP addresses as below:

 in the PC :   192.168.10.127   and  additional one for  RSI interface: 192.168.1.30 
Controller side:   windows interface (KLI) : 192.168.10.10  and the RSI interface 192.168.1.20
(all with net mask 255.255.255.0)

in `ros_rsi_ethernet.xml   , IP_NUMBER and PORT` are set as `192.168.1.30` and `49152`

I managed to ping `192.168.10.127` from the controller. 

in addition, I have modified the provided `kr6r900sixx_macro.xacro` file to achieve a robot model in default formation. 

after running `roslaunch kuka_rsi_hw_interface test_hardware_interface.launch sim:=false`
```
`Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://stamPC:46730/

PARAMETERS
 * /controller_joint_names: ['joint_a1', 'joi...
 * /joint_state_controller/publish_rate: 50
 * /joint_state_controller/type: joint_state_contr...
 * /position_trajectory_controller/action_monitor_rate: 20
 * /position_trajectory_controller/joints: ['joint_a1', 'joi...
 * /position_trajectory_controller/state_publish_rate: 50
 * /position_trajectory_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /rosdistro: indigo
 * /rosversion: 1.11.21
 * /rsi/listen_address: 192.168.1.30
 * /rsi/listen_port: 49152

`NODES`

    controller_spawner (controller_manager/spawner)
    kuka_hardware_interface (kuka_rsi_hw_interface/kuka_hardware_interface_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

`ROS_MASTER_URI=http://localhost:11311`

core service [/rosout] found
process[kuka_hardware_interface-1]: started with pid [22881]
process[controller_spawner-2]: started with pid [22882]
[ INFO] [1520255821.872543849]: Starting hardware interface...
process[robot_state_publisher-3]: started with pid [22883]
[ INFO] [1520255821.915341175]: Loaded kuka_rsi_hardware_interface
[ INFO] [1520255821.918251058]: Setting up RSI server on: (192.168.1.30, 49152)
[ INFO] [1520255822.064710726]: Waiting for robot!
[INFO] [WallTime: 1520255822.752157] Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [WallTime: 1520255822.754764] Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [WallTime: 1520255822.757702] Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [WallTime: 1520255822.760265] Loading controller: position_trajectory_controller
[INFO] [WallTime: 1520255822.831506] Loading controller: joint_state_controller
[INFO] [WallTime: 1520255822.838806] Controller Spawner: Loaded controllers: position_trajectory_controller, joint_state_controller
`
```
**When I try to run the ROS_RSI KRL file, after the second step (sensor correction is activated), I receive the below Error on the teach-pendent:** 

1 Block change on STOP
Originator: /R1/ROS_RSI  

2 Stop by $CORRECTION function ( reset or block selection required) 
Originator: A2

3 Signal flow(running): Object ETHERNET1 returns error RSITimeout

gavanderhoorn, 2018-03-05:
@monykhosrvi wrote:
> in addition, I have modified the provided `kr6r900sixx_macro.xacro` file to achieve a robot model in default formation. 

can you explain what you mean by this?

Editing the robot support package should not be necessary, unless you have a very customised setup.

monykhosrvi, 2018-03-05:
@gavanderhoorn thanks for reply,   

I tried to set-up an RVIZ package with the .xacro file stored here [ file](https://github.com/ros-industrial/kuka_experimental/blob/indigo-devel/kuka_kr6_support/urdf/kr6r900sixx_macro.xacro) . 

and I ended up with a robot model fully stretched  ( all joint rpy set at 0 0 0 . example " `<origin xyz="0.080 0 0" rpy="0 0 0"/>`).

I have already read the issue #84 . and I thiught I may have the same issue so I adjusted the rpy definitions to achieve a formation equal to the  BCO. 

That was the only change I made.

gavanderhoorn, 2018-03-05:
@monykhosrvi wrote:
> and I ended up with a robot model fully stretched  ( all joint rpy set at 0 0 0 . example " `<origin xyz="0.080 0 0" rpy="0 0 0"/>`).

that is expected: all joints at 0 should result in an arm 'fully stretched'.

monykhosrvi, 2018-03-05:
@gavanderhoorn  

Do you think the mentioned problems:

1 Block change on STOP
Originator: /R1/ROS_RSI  

2 Stop by $CORRECTION function ( reset or block selection required) 
Originator: A2

3 Signal flow(running): Object ETHERNET1 returns error RSITimeout

are due to not having RT-Preempt kernel installed ?

gavanderhoorn, 2018-03-05:
I only see a `[ INFO] [1520255822.064710726]: Waiting for robot!` in your log output. If the robot actually connects to your PC, I expect to also see a `Got connection from robot` ([here](https://github.com/ros-industrial/kuka_experimental/blob/f84604221a8738f2e06d2268aed505a03d0c6537/kuka_rsi_hw_interface/src/kuka_hardware_interface.cpp#L158)).

It could be a timeout, but that would be the even before the PC and the controller have succesfully exchanged the fitst `RSICommand`. Can you make sure IP connectivity is correctly configured?

monykhosrvi, 2018-03-05:
>  Can you make sure IP connectivity is correctly configured?

honestly that is my main concern, in the instruction:

> Windows runs behind the SmartHMI on the teach pad. Make sure that the Windows interface of the controller and the PC with ROS is connected to the same subnet.

the windows network setting on the controller  are 192.168.0.1  255.255.255.0 once i tried to change it (from `'Control Panel\Network and Internet\Network Connections'`) and i experienced fatal problem with the teach pendent connection. so i had to  restore the Robot from the backup image. 

considering the already explained:

>I have set the IP addresses as below: in the PC with ROS : 192.168.10.127 and additional one for RSI interface: 192.168.1.30
Controller side: windows interface (KLI) : 192.168.10.10 and the RSI interface 192.168.1.20
(all with net mask 255.255.255.0)

It is not clear for me if the windows interface (KLI) visible in the (Start-up > Network configuration > Activate advanced configuration).  is the same which is named as "Windows interface of the controller" in the instruction or not ? 

in addition , after adding a new IP address on another subnet (e.g. 192.168.1.20) for the RSI interface and performing a cold start, shall we check it out as the Windows interface in the Network advanced configuration or not?

gavanderhoorn, 2018-03-05:
@tingelst, @ivareri, @BrettHemes: any ideas?

BrettHemes, 2018-03-08:
> I only see a [ INFO] [1520255822.064710726]: Waiting for robot! in your log output. If the robot actually connects to your PC, I expect to also see a Got connection from robot (here).

Yes, it seems the connection is never happening and the RSI interface on the robot controller is timing out due to unanswered transmissions.

> the windows network setting on the controller are 192.168.0.1 255.255.255.0 once i tried to change it (from 'Control Panel\Network and Internet\Network Connections') and i experienced fatal problem with the teach pendent connection. so i had to restore the Robot from the backup image.

Ahhh yes, don't change network settings from the Windows side unless you really know what you are doing.  A wrong setting here destroys connections between Windows/RTOS/HMI... not good.  Seems you learned this one the hard way...

**Here are some pictures of one of my KR6's with RSI/ROS setup that hopefully provide a benchmark for comparison:**
This is the Basic (i.e., not Advanced) configuration view and it shows the Windows interface (virtual6).  You can use this to ping the controller and connect via WorkVisual etc. but it isn't really important for the RSI setup (just a sanity check).
[image]

Note the `Windows interface` check box of the same `virtual6` interface on the Advanced view:
[image]

`virtual5`... (labeled KLI on mine?) It is not relevant for RSI setup and shouldn't be messed with generally (see note above):
[image]

And finally my RSI interface (`virtual7`).  This is the important one and needs to be on the same subnet as your ROS PC's RSI adapter:
[image]

monykhosrvi, 2018-03-08:
@BrettHemes  I do appreciate your remarks.  seems that I  misinterpreted the windows interface with the KLI one.  Tomorrow I will set up the network setting as described and update the issue.

monykhosrvi, 2018-03-09:
I have modified the controller's network setting as:
`virtual7`  no changes
`virtual6`  Windows interface   `192.168.10.10`
`virtual8`   RSI                          `192.168.1.20`

on the PC with ROS:

```
$ hostname -I
192.168.10.127 192.168.1.30 10.99.157.16 
```

I can ping 192.168.10.10 from PC and 192.168.10.127 from the controller but I can not ping 192.168.1.30 from KRC nor 192.168.1.20 from PC. 

In ros_rsi_ethernet.xml  the IP address is is set at 192.168.1.30.

 I  receive the same error after a second. 
1 Block change on STOP
Originator: /R1/ROS_RSI  

2 Stop by $CORRECTION function ( reset or block selection required) 
Originator: A2

3 Signal flow(running): Object ETHERNET1 returns error RSITimeout

since the ROS PC is connected to an Hyper-spectral camera which required also a setting of  DNS address,   I suspected that might be the problem. 

then I tried to configure the PC network setting with different [approach](https://askubuntu.com/questions/313877/how-do-i-add-an-additional-ip-address-to-etc-network-interfacesl)

and finally I managed to ping the RSI interface from the PC and establish the RSI connection. 

no I am facing the the problem of RSI shutting down while robot moving foR which I will compile and install RT-Preempt kernel. 

@gavanderhoorn ,@BrettHemes ,  at the end I would like to thank you both for your answers and help.

gavanderhoorn, 2018-03-09:
Ok, no problem. Glad to hear you got the initial configuration sorted out.

@BrettHemes: we might want to think about adding some diagrams or screenshots to the readme / installation and setup documentation, just to avoid confusion in the future.

gavanderhoorn, 2018-03-09:
@monykhosrvi: I'm going to close this issue then as it seems you have resolved it, and it's not really a problem with any of the packages in this repository.

Feel free to keep commenting though.

closed 2018-03-09 by gavanderhoorn