ros-industrial/kuka_experimental #122: Error Axiscorrmon1 returns error RSIInvalidParam

masterage2014, 2018-04-06:
Hi,
i am new to ROS. 

I tried connecting the robot via rsi according to: https://github.com/ros-industrial/kuka_experimental/tree/indigo-devel/kuka_rsi_hw_interface/krl/KR_C4
 
When i come to point 4.4 (Testing) i get the following error message on the smartPad:

******
x) Stop by $Correction function (reset or block selection requiered) Originator: A2
x) Signal flow (init): Object AXISCORRMON1 returns error RSIInvalidParam
******

Any ideas, what i could have done wrong ?

With best regards!!

gavanderhoorn, 2018-04-06:
I believe we're first going to need some more information: do you have a KR C4 controller? Which version of RSI do you have installed? What is the version of KSS?

masterage2014, 2018-04-07:
Hi, 

thank you for the fast response. We solved the problem by re-installing the rsi to the controller.

Best regards !

closed 2018-04-07 by masterage2014