ros-industrial/kuka_experimental #123: "Failed to read state from robot. Shutting down!" between two movements

cschindlbeck, 2018-04-13:
I need to do some calculations between two robot movements (i.e. two consecutive movements are timely seperated by a couple of minutes). The ROS interface shuts down after a while with "Failed to read state from robot. Shutting down!" so that i am forced to run ros_rsi from the beginning.
I guess this is intended and i shouldn't just uncomment this line in the kuka_hardware_interface_node.cpp.
What would be the best course of action here?

gavanderhoorn, 2018-04-13:
Without more information this is going to be difficult to diagnose.

Have you always run Kinetic with this setup? Has anything changed on the robot side? Any MoveIt updates that you performed? Have you tried turning the controller on and off (you'd be surprised)?

Any hardware changes? Using wireless instead of wired network by mistake? etc

These kind of errors point to missing/dropped packets, problems with trajectories (violations of limits) or something else.

If you can, plot some trajectories and see what is going on.

BrettHemes, 2018-04-14:
I assuming this is with `kuka_rsi_hw_interface`?

cschindlbeck, 2018-04-16:
Sorry, i will try to be more verbose.

> I assuming this is with kuka_rsi_hw_interface?
Yes.

I am pretty sure now this is unrelated to my switch to kinetic.
I have connected Gazebo to efficiently debug my problem (see PR). The trajectories are really smooth and i even slowed them down to be sure that i do not run into some velocity limits. 
  

> (EEF changed without updating payload settings)

Yeah i actually attached a small aluminum plate. Since it is lightweight (it has an off-axis center of mass though) i thought this cannot be the problem. Giving some thought to your comment this is the only thing that makes sense. How can i change the payload settings?

gavanderhoorn, 2018-04-16:
Note that I was just guessing. The payload settings could be a red herring.

re: updating the Troubleshooting guide might make sense, if it turns out to be the payload.

cschindlbeck, 2018-04-17:
This issue was removed by:

- Accounting for the payload in the KUKA controller
- Activation of HOLDON on the KUKA C4, see https://github.com/ros-industrial/kuka_experimental/pull/119

Thanks for the tip @gavanderhoorn

closed 2018-04-17 by cschindlbeck

RhysMcK, 2018-04-17:
@cschindlbeck how accurately did you have to set the payload in the KUKA controller? Did you use the KUKA provided software- Kuka.LoadDataDetermination, or did you manually add the mass, CoM and inertia's. Thanks!

cschindlbeck, 2018-04-18:
I have a custom gripper made of aluminum and an aluminum plate so i have inferred the inertial data from Solidworks (keep in mind to export your data wrt to the correct coordinate system). The only thing not accounted for is a servo motor so it's not totally spot on but i would say 95% accuracy. I entered the data manually with the teach pendant (Calibration -> Tool -> load supplementary data). 
I forgot to mention that i have furthermore slowed down the trajectories with
`
set_max_velocity_scaling_factor(0.02)
set_max_acceleration_scaling_factor(0.01)
`

I haven't sped up the trajectories to "normal" speed yet. Maybe you can try with really slow trajectory execution and work you way from there.

RhysMcK, 2018-04-19:
Thanks @cschindlbeck for getting in touch. If your using a similar environment setup to myself, i fear you might once again encounter this error when you speed things back up to "normal" speed.   Please see my lasted comment on this discuss  https://github.com/ros-industrial/kuka_experimental/issues/126#issuecomment-382572199 . Cheers!