ros-industrial/kuka_experimental #129: kr5 effort limit?

CesMak, 2018-04-27:
Hey there,

I just loaded the kr5_arc robot description in gazebo and setup a ros_controller. Now if I just unpause the gazebo simulator the whole kr5 robot collapses instantly. That is why I set in the .urdf file the  limit effort instead of 0 to 10.

I basically did that for all joints but actually I think that this is not correct .... 

```

<joint name="${prefix}joint_a1" type="revolute">
  <origin xyz="0 0 0.4" rpy="${pi} 0 0"/>
  <parent link="${prefix}base_link"/>
  <child link="${prefix}link_1"/>
  <axis xyz="0 0 1"/>
  <limit effort="${joint_effort_limit_1}" lower="${radians(-155)}" upper="${radians(155)}" velocity="${radians(154)}"/>
</joint>
```

Moreover I wonder how  to setup the p, i, d  values of the controller.

Currently I have:

```

manipulator:
  # Publish all joint states -----------------------------------
  joint_state_controller:
    type: joint_state_controller/JointStateController
    publish_rate: 50  
  
  # Position Controllers ---------------------------------------
  link_1_controller:
    type: effort_controllers/JointPositionController
    joint: joint_a1
    pid: {p: 1000.00, i: 200.00, d: 200.00}

  link_2_controller:
    type: effort_controllers/JointPositionController
    joint: joint_a2
    pid: {p: 5000.00, i: 500.00, d: 500.00}

  link_3_controller:
    type: effort_controllers/JointPositionController
    joint: joint_a3
    pid: {p: 1000.00, i: 750.00, d: 50.00}

  link_4_controller:
    type: effort_controllers/JointPositionController
    joint: joint_a4
    pid: {p: 1000.00, i: 200.00, d: 200.00}

  link_5_controller:
    type: effort_controllers/JointPositionController
    joint: joint_a5
    pid: {p: 5000.00, i: 500.00, d: 500.00}

  link_6_controller:
    type: effort_controllers/JointPositionController
    joint: joint_a6
    pid: {p: 1000.00, i: 750.00, d: 50.00}
```

Any links or explanation how to set the **effort limits and the pid** value would be great!

gavanderhoorn, 2018-04-27:
There is no effort limit set in the xacro macro as we don't know what it should be. KUKA does not give out that sort of information without NDA.

As to the controller configuration: I typically use position controllers to avoid having to tune the PIDs.

In any case, as this is a really per-model and per-setup task, I'm afraid I can't be of any help.

I would also recommend to ask this sort of question on ROS Answers, as we'd like to keep the issue tracker for enhancement requests and real problems with the packages in this repository.

gavanderhoorn, 2018-05-02:
Is [choose right ros_control configuration for a kuka arm](https://answers.ros.org/question/290181) you on ROS Answers?

CesMak, 2018-05-02:
thanks I asked my question here:
https://answers.ros.org/question/290181/choose-right-ros_control-configuration-for-a-kuka-arm/

gavanderhoorn, 2018-05-03:
Then I'll close this one.

closed 2018-05-03 by gavanderhoorn