ros-industrial/kuka_experimental #130: kr5_arc construction dh table

CesMak, 2018-05-03:
Hey there,

I would like to know the dh table of the kr5_arc. What I found here:
https://github.com/petercorke/robotics-toolbox-matlab/blob/master/mdl_KR5.m

was: 
%            theta    d           a       alpha
L(1) = Link([0        0.4         0.18    pi/2]);
L(2) = Link([0        0.135       0.60    pi]);
L(3) = Link([0        0.135       0.12   -pi/2]);
L(4) = Link([0        0.62        0       pi/2]);
L(5) = Link([0        0           0      -pi/2]);
L(6) = Link([0 0 0 0]);

But I think this kuka_kr5_arc does not have the 0.135 in link2 and link3 or? 

```
      <!-- joint 1 (A1) -->
      <joint name="${prefix}joint_a1" type="revolute">
         <origin xyz="0 0 0.4" rpy="${pi} 0 0" />
         <parent link="${prefix}base_link" />
         <child link="${prefix}link_1" />
         <axis xyz="0 0 1" />
         <limit effort="${joint_effort_limit_1}" lower="${radians(-155)}" upper="${radians(155)}" velocity="${radians(154)}" />
      </joint>
      <!-- joint 2 (A2) -->
      <joint name="${prefix}joint_a2" type="revolute">
         <origin xyz="0.18 0 0" rpy="${pi/2} 0.0 0" /> <!-- 2.4 is my value (second) -->
         <parent link="${prefix}link_1" />
         <child link="${prefix}link_2" />
         <axis xyz="0 0 1" />
         <limit effort="${joint_effort_limit_2}" lower="${radians(-180)}" upper="${radians(65)}" velocity="${radians(154)}" />
      </joint>
      <!-- joint 3 (A3) -->
      <joint name="${prefix}joint_a3" type="revolute">
         <origin xyz="0.6 0 0" rpy="0 0 0" /> <!-- 1.5 is my value (third) -->
         <parent link="${prefix}link_2" />
         <child link="${prefix}link_3" />
         <axis xyz="0 0 1" />
         <limit effort="${joint_effort_limit_3}" lower="${radians(-15)}" upper="${radians(158)}" velocity="${radians(228)}" />
      </joint>
      <!-- joint 4 (A4) -->
      <joint name="${prefix}joint_a4" type="revolute">
         <origin xyz="0 -0.12 0" rpy="0 ${-pi/2} 0" />
         <parent link="${prefix}link_3" />
         <child link="${prefix}link_4" />
         <axis xyz="0 0 1" />
         <limit effort="${joint_effort_limit_4}" lower="${radians(-350)}" upper="${radians(350)}" velocity="100" />
         <!-- ${radians(343)}-->
      </joint>
      <!-- joint 5 (A5) -->
      <joint name="${prefix}joint_a5" type="revolute">
         <origin xyz="0 0 -0.62" rpy="0 ${pi/2} 0" />
         <parent link="${prefix}link_4" />
         <child link="${prefix}link_5" />
         <axis xyz="0 0 1" />
         <limit effort="${joint_effort_limit_5}" lower="${radians(-130)}" upper="${radians(130)}" velocity="100" />
         <!-- ${radians(384)}-->
      </joint>
      <!-- joint 6 (A6) -->
      <joint name="${prefix}joint_a6" type="revolute">
         <origin xyz="0 0 0" rpy="0 ${-pi/2} 0" />
         <parent link="${prefix}link_5" />
         <child link="${prefix}link_6" />
         <axis xyz="0 0 1" />
         <limit effort="${joint_effort_limit_6}" lower="${radians(-350)}" upper="${radians(350)}" velocity="100" />
         <!--${radians(721)} -->
      </joint>
      <!-- tool frame - fixed frame -->
      <joint name="${prefix}joint_a6-tool0" type="fixed">
         <parent link="${prefix}link_6" />
         <child link="${prefix}tool0" />
         <origin xyz="0 0 -0.115" rpy="0 ${-pi} ${-pi/2}" />
      </joint>
```

gavanderhoorn, 2018-05-03:
I'm not entirely sure what you are asking.

The xacro macro was not necessarily constructed from DH parameters, nor are they used in ROS.

We'd have to ask @destogl (who contributed it in #79) where he got those values.

Personally I always use the diagrams provided by the manufacturer (shown in https://github.com/ros-industrial/kuka_experimental/pull/79#issuecomment-308380198).

BrettHemes, 2018-05-20:
The DH parameters, if needed, should be straightforwardly derivable from the information provided in the manuals and/or available CAD models.  However, as @gavanderhoorn they will not necessarily correspond to the values in the urdf's.  I checked the kr5_arc_macro.xacro with the Kuka manual and everything is in agreement. 

Closing this as it is not an issue.  Please post elsewhere for general questions (i.e., help with DH).

closed 2018-05-20 by BrettHemes