ros-industrial/kuka_experimental #131: ROS and KUKA KR120 r2700 EXTRA HA

romainbaisse, 2018-05-17:
Hi everyone, 
I am a French student currently doing an internship in University of Technology Sydney. I have to add a 3D visual feedback to a Kuka robot used for 3D printing. The model of the robot is KR120 R2700 EXTRA HA (with KRC4). No one seems to use this robot on the Internet. So here I am. 

I have to connect this robot with ROS, but there is not support yet for this robot in the kuka_experimental package. Is there someone who have ideas of how I can do that ? 
No .urdf or .xacro file exist for this robot, and my internship supervisor don't want me to do that by myself (because too dangerous seeing the size of the robot and the configuration of the room the robot is in). 

Really thank you for your answer.
(and sorry for my bad english ahah)

romainbaisse, 2018-05-17:
I don’t want that someone else do that for me I am just asking if someone knows another way to connect this robot with ROS without an urdf file.
I am new with ROS so I didn’t know so much think about it.

romainbaisse, 2018-05-17:
Yes sorry I will try to explain with more details. 
I reformulate my problem : 
I have to acces to the robot state (joints values) ans also to send velocity command from my computer to the robot through ROS. How can I connect to the robot ? 
In the README.md of the kuka_rsi_hd_interface it's written in part 3 that I have to add the robot_description in order to successfully launch the kuka_rsi_hd_interface. 
So as I said before I haven't got the file needed for my robot. So i don't know how I can communicate with the robot. 

Hope it is clear.

gavanderhoorn, 2018-05-17:
> I have to acces to the robot state (joints values) 

ok.

> ans also to send velocity command from my computer to the robot through ROS. 

Then unfortunately you cannot use the RSI or EKI components we have in this repository. Nor RSI, nor EKI supports any velocity interfaces, so what you're after isn't possible.

Afaik KUKA has no velocity control interfaces available for KR C4 controlled robots (but I could be wrong).

BrettHemes, 2018-05-20:
> > I don’t want that someone else do that for me I am just asking if someone knows another way to connect this robot with ROS without an urdf file.

> a urdf is not needed to connect with your robot.

> both RSI and EKI can be used without a urdf

>> ans also to send velocity command from my computer to the robot through ROS.

> Then unfortunately you cannot use the RSI or EKI components we have in this repository. Nor RSI, nor EKI supports any velocity interfaces, so what you're after isn't possible.

Can we close this?

romainbaisse, 2018-05-20:
Yes sorry I have been busy.

We can close it !

thank you for your answer.

closed 2018-05-20 by BrettHemes

gavanderhoorn, 2018-05-21:
So what did you end up doing @romainbaisse? Did you find a velocity control interface somewhere for RSI, or did you end up with position control after all?

romainbaisse, 2018-05-21:
Currently I am not working on that part, I am more focused on the visual part of my project. But surely I will go for position control when I will working back on the robot. 
Thanks.

BrettHemes, 2018-05-22:
> This is not a ROS limitation, it's just that afaik KUKA has no velocity control interfaces available for KR C4 controlled robots (but I could be wrong).

@gavanderhoorn I would be curious how well we could approximate such an interface in RSI

gavanderhoorn, 2018-05-22:
@BrettHemes: we could probably implement something of a poor-mans velocity controller. Without deterministic scheduling though, I'm not sure how nice the performance would be.