ros-industrial/kuka_experimental #133: How To launch multiple arms?

skyrobos, 2018-06-10:
Hi,
I had tried to launch two arms with no success.
This is  the launch file:
````

<?xml version="1.0" encoding="utf-8"?>
<launch>
  <arg name="sim" default="true" />

  <group ns="ag1">
    <param name="robot_description" command="$(find xacro)/xacro.py '$(find kuka_kr10_support)/urdf/kr10r1100sixx_macro.xacro'"/>
    <!-- Load RSI parameters -->
    <rosparam file="$(find kuka_rsi_hw_interface)/test/ag1_params.yaml" command="load" />
    <!-- Start node -->
    <node name="kuka_hardware_interface" pkg="kuka_rsi_hw_interface"
          type="kuka_hardware_interface_node" respawn="false"
          output="screen"
          required="true"/>
    <!-- Load joint controller configurations from YAML file to parameter server -->
    <rosparam file="$(find kuka_rsi_hw_interface)/test/ag1_hw_controllers.yaml" command="load"/>
    <!-- Load standard kuka controller joint names from YAML file to parameter server -->
    <rosparam file="$(find kuka_rsi_hw_interface)/test/ag1_controller_joint_names.yaml" command="load"/>
    <!-- Load controllers -->
    <node name="controller_spawner" pkg="controller_manager" type="spawner" respawn="false" output="screen"
        args="position_trajectory_controller joint_state_controller --shutdown-timeout 1"/>
    <!-- Load robot state publisher -->
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />
    <!-- Load RSI simulation node -->
    <node name='kuka_rsi_simulator' pkg='kuka_rsi_simulator' type="kuka_rsi_simulator" args="127.0.0.1 49153" if="$(arg sim)" />
    </group>

  <group ns="ag2">
    <param name="robot_description" command="$(find xacro)/xacro.py '$(find kuka_kr10_support)/urdf/kr10r1100sixx_macro.xacro'"/>
    <!-- Load RSI parameters -->
    <rosparam file="$(find kuka_rsi_hw_interface)/test/ag2_params.yaml" command="load" />
    <!-- Start node -->
    <node name="kuka_hardware_interface" pkg="kuka_rsi_hw_interface"
          type="kuka_hardware_interface_node" respawn="false"
          output="screen"
          required="true"/>
    <!-- Load joint controller configurations from YAML file to parameter server -->
    <rosparam file="$(find kuka_rsi_hw_interface)/test/ag2_hw_controllers.yaml" command="load"/>
    <!-- Load standard kuka controller joint names from YAML file to parameter server -->
    <rosparam file="$(find kuka_rsi_hw_interface)/test/ag2_controller_joint_names.yaml" command="load"/>
    <!-- Load controllers -->
    <node name="controller_spawner" pkg="controller_manager" type="spawner" respawn="false" output="screen"
        args="position_trajectory_controller joint_state_controller --shutdown-timeout 1"/>
    <!-- Load robot state publisher -->
    <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />
    <!-- Load RSI simulation node -->
    <node name='kuka_rsi_simulator' pkg='kuka_rsi_simulator' type="kuka_rsi_simulator" args="127.0.0.1 49152" if="$(arg sim)" />
    </group>

</launch>
````

Error message:
````

[ag1/robot_state_publisher-3] process has died [pid 22119, exit code 255, cmd /opt/ros/kinetic/lib/robot_state_publisher/robot_state_publisher /follow_joint_trajectory:=position_trajectory_controller/follow_joint_trajectory __name:=robot_state_publisher __log:=/home/roblabo/.ros/log/38d8bf14-6cb5-11e8-9fa2-0028f84e1457/ag1-robot_state_publisher-3.log].
log file: /home/roblabo/.ros/log/38d8bf14-6cb5-11e8-9fa2-0028f84e1457/ag1-robot_state_publisher-3*.log
````

What did I do wrong?

gavanderhoorn, 2018-06-12:
@skyrobos wrote:
> What did I do wrong?

I can't say. Or at least not from the text you post.

Can you start your launch file using `roslaunch --screen ..` and post the complete output here?

skyrobos, 2018-06-12:
````
xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://ros.org/wiki/xacro
 new: http://wiki.ros.org/xacro (/home/roblabo/catkin_ws/src/kuka_experimental/kuka_resources/urdf/common_constants.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://ros.org/wiki/xacro
 new: http://wiki.ros.org/xacro (/home/roblabo/catkin_ws/src/kuka_experimental/kuka_resources/urdf/common_materials.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://www.ros.org/wiki/xacro
 new: http://ros.org/wiki/xacro (/home/roblabo/catkin_ws/src/kuka_experimental/kuka_kr10_support/urdf/kr10r1100sixx_macro.xacro)
xacro.py is deprecated; please use xacro instead
xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://ros.org/wiki/xacro
 new: http://wiki.ros.org/xacro (/home/roblabo/catkin_ws/src/kuka_experimental/kuka_resources/urdf/common_constants.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://ros.org/wiki/xacro
 new: http://wiki.ros.org/xacro (/home/roblabo/catkin_ws/src/kuka_experimental/kuka_resources/urdf/common_materials.xacro)
xacro.py is deprecated; please use xacro instead
xacro: Traditional processing is deprecated. Switch to --inorder processing!
To check for compatibility of your document, use option --check-order.
For more infos, see http://wiki.ros.org/xacro#Processing_Order
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://ros.org/wiki/xacro
 new: http://wiki.ros.org/xacro (/home/roblabo/catkin_ws/src/kuka_experimental/kuka_resources/urdf/common_constants.xacro)
inconsistent namespace redefinitions for xmlns:xacro:
 old: http://ros.org/wiki/xacro
 new: http://wiki.ros.org/xacro (/home/roblabo/catkin_ws/src/kuka_experimental/kuka_resources/urdf/common_materials.xacro)
xacro.py is deprecated; please use xacro instead
started roslaunch server http://roblabo:34043/

SUMMARY
========

PARAMETERS
 * /ag1/controller_joint_names: ['ag1_joint_1', '...
 * /ag1/joint_state_controller/publish_rate: 50
 * /ag1/joint_state_controller/type: joint_state_contr...
 * /ag1/position_trajectory_controller/action_monitor_rate: 20
 * /ag1/position_trajectory_controller/joints: ['ag1_joint_1', '...
 * /ag1/position_trajectory_controller/state_publish_rate: 50
 * /ag1/position_trajectory_controller/type: position_controll...
 * /ag1/robot_description: <?xml version="1....
 * /ag1/rsi/listen_address: 192.168.1.11
 * /ag1/rsi/listen_port: 49151
 * /ag2/controller_joint_names: ['ag2_joint_1', '...
 * /ag2/joint_state_controller/publish_rate: 50
 * /ag2/joint_state_controller/type: joint_state_contr...
 * /ag2/position_trajectory_controller/action_monitor_rate: 20
 * /ag2/position_trajectory_controller/joints: ['ag2_joint_1', '...
 * /ag2/position_trajectory_controller/state_publish_rate: 50
 * /ag2/position_trajectory_controller/type: position_controll...
 * /ag2/robot_description: <?xml version="1....
 * /ag2/rsi/listen_address: 192.168.1.11
 * /ag2/rsi/listen_port: 49152
 * /controller_joint_names: ['joint_a1', 'joi...
 * /move_group/allow_trajectory_execution: True
 * /move_group/capabilities: move_group/MoveGr...
 * /move_group/controller_list: [{'action_ns': 'f...
 * /move_group/endeffector/planner_configs: ['SBLkConfigDefau...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/manipulator/longest_valid_segment_fraction: 0.005
 * /move_group/manipulator/planner_configs: ['SBLkConfigDefau...
 * /move_group/manipulator/projection_evaluator: joints(joint_a1,j...
 * /move_group/max_range: 5.0
[... 238 lines cut ...]
Traceback (most recent call last):
  File "/home/roblabo/catkin_ws/src/kuka_experimental/kuka_rsi_simulator/scripts/kuka_rsi_simulator", line 60, in <module>
    rospy.init_node(node_name)
  File "/opt/ros/kinetic/lib/python2.7/dist-packages/rospy/client.py", line 323, in init_node
    raise rospy.exceptions.ROSInitException("init_node interrupted before it could complete")
rospy.exceptions.ROSInitException: init_node interrupted before it could complete
Loading 'move_group/ApplyPlanningSceneService'...
Loading 'move_group/ClearOctomapService'...
Loading 'move_group/MoveGroupCartesianPathService'...
Loading 'move_group/MoveGroupExecuteTrajectoryAction'...
Loading 'move_group/MoveGroupGetPlanningSceneService'...
Loading 'move_group/MoveGroupKinematicsService'...
Loading 'move_group/MoveGroupMoveAction'...
Loading 'move_group/MoveGroupPickPlaceAction'...
Loading 'move_group/MoveGroupPlanService'...
Loading 'move_group/MoveGroupQueryPlannersService'...
Loading 'move_group/MoveGroupStateValidationService'...

You can start planning now!

shutting down processing monitor...
... shutting down processing monitor complete
done

````

skyrobos, 2018-06-12:
I did have a problem with the IP configuration.

Unfortunately now It throws other errors:
````
SUMMARY
========

PARAMETERS
 * /ag1/controller_joint_names: ['ag1_joint_1', '...
 * /ag1/joint_state_controller/publish_rate: 50
 * /ag1/joint_state_controller/type: joint_state_contr...
 * /ag1/position_trajectory_controller/action_monitor_rate: 20
 * /ag1/position_trajectory_controller/joints: ['ag1_joint_1', '...
 * /ag1/position_trajectory_controller/state_publish_rate: 50
 * /ag1/position_trajectory_controller/type: position_controll...
 * /ag1/robot_description: <?xml version="1....
 * /ag1/rsi/listen_address: 127.0.0.1
 * /ag1/rsi/listen_port: 49151
 * /ag2/controller_joint_names: ['ag2_joint_1', '...
 * /ag2/joint_state_controller/publish_rate: 50
 * /ag2/joint_state_controller/type: joint_state_contr...
 * /ag2/position_trajectory_controller/action_monitor_rate: 20
 * /ag2/position_trajectory_controller/joints: ['ag2_joint_1', '...
 * /ag2/position_trajectory_controller/state_publish_rate: 50
 * /ag2/position_trajectory_controller/type: position_controll...
 * /ag2/robot_description: <?xml version="1....
 * /ag2/rsi/listen_address: 127.0.0.1
 * /ag2/rsi/listen_port: 49152
 * /controller_joint_names: ['joint_a1', 'joi...
 * /move_group/allow_trajectory_execution: True
 * /move_group/capabilities: move_group/MoveGr...
 * /move_group/controller_list: [{'action_ns': 'f...
 * /move_group/endeffector/planner_configs: ['SBLkConfigDefau...
 * /move_group/jiggle_fraction: 0.05
 * /move_group/manipulator/longest_valid_segment_fraction: 0.005
 * /move_group/manipulator/planner_configs: ['SBLkConfigDefau...
 * /move_group/manipulator/projection_evaluator: joints(joint_a1,j...
 * /move_group/max_range: 5.0
 * /move_group/max_safe_path_cost: 1
 * /move_group/moveit_controller_manager: moveit_simple_con...
 * /move_group/moveit_manage_controllers: True
 * /move_group/octomap_resolution: 0.025
 * /move_group/planner_configs/BFMTkConfigDefault/balanced: 0
 * /move_group/planner_configs/BFMTkConfigDefault/cache_cc: 1
 * /move_group/planner_configs/BFMTkConfigDefault/extended_fmt: 1
 * /move_group/planner_configs/BFMTkConfigDefault/heuristics: 1
 * /move_group/planner_configs/BFMTkConfigDefault/nearest_k: 1
 * /move_group/planner_configs/BFMTkConfigDefault/num_samples: 1000
 * /move_group/planner_configs/BFMTkConfigDefault/optimality: 1
 * /move_group/planner_configs/BFMTkConfigDefault/radius_multiplier: 1.0
 * /move_group/planner_configs/BFMTkConfigDefault/type: geometric::BFMT
 * /move_group/planner_configs/BKPIECEkConfigDefault/border_fraction: 0.9
 * /move_group/planner_configs/BKPIECEkConfigDefault/failed_expansion_score_factor: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/min_valid_path_fraction: 0.5
 * /move_group/planner_configs/BKPIECEkConfigDefault/range: 0.0
 * /move_group/planner_configs/BKPIECEkConfigDefault/type: geometric::BKPIECE
 * /move_group/planner_configs/BiESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/BiESTkConfigDefault/type: geometric::BiEST
 * /move_group/planner_configs/BiTRRTkConfigDefault/cost_threshold: 1e300
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_node_ratio: 0.1
 * /move_group/planner_configs/BiTRRTkConfigDefault/frountier_threshold: 0.0
 * /move_group/planner_configs/BiTRRTkConfigDefault/init_temperature: 100
 * /move_group/planner_configs/BiTRRTkConfigDefault/range: 0.0
 * /move_group/planner_configs/BiTRRTkConfigDefault/temp_change_factor: 0.1
 * /move_group/planner_configs/BiTRRTkConfigDefault/type: geometric::BiTRRT
 * /move_group/planner_configs/ESTkConfigDefault/goal_bias: 0.05
 * /move_group/planner_configs/ESTkConfigDefault/range: 0.0
 * /move_group/planner_configs/ESTkConfigDefault/type: geometric::EST
 * /move_group/planner_configs/FMTkConfigDefault/cache_cc: 1
 * /move_group/planner_configs/FMTkConfigDefault/extended_fmt: 1
[... 247 lines cut ...]
*     - ApplyPlanningSceneService
*     - ClearOctomapService
*     - CartesianPathService
*     - ExecuteTrajectoryAction
*     - GetPlanningSceneService
*     - KinematicsService
*     - MoveAction
*     - PickPlaceAction
*     - MotionPlanService
*     - QueryPlannersService
*     - StateValidationService
********************************************************

[ INFO] [1528811564.225908989]: MoveGroup context using planning plugin ompl_interface/OMPLPlanner
[ INFO] [1528811564.225930856]: MoveGroup context initialization complete

You can start planning now!

````
It cant load the URDF, but when I launch one arm with the test_hardware_interface.launch all works, so it's not a URDF problem.

I use that command in my launch file (the full launch configuration is in the first comment):
````
    <param name="robot_description" command="$(find xacro)/xacro.py '$(find kuka_kr10_support)/urdf/kr10r1100sixx_macro.xacro'"/>
````

gavanderhoorn, 2018-06-12:
You're loading the wrong file. `_macro.xacro` files load the macro, they're not top-level xacros.

Load the `.xacro` file.

But do so by reusing the `launch/load_kr10r1100sixx.launch` (as explained [here](http://wiki.ros.org/Industrial/Tutorials/WorkingWithRosIndustrialRobotSupportPackages)).

skyrobos, 2018-06-12:
Thank you for the help!
I do have some other related issues:
1. I need to remap the follow_joint_trajectory to the relevant namespace/position_trajectory_controller.
The problem is that I cant use the remap at the moveit configuration because I have two instances of follow_joint_trajectory topic. I tried to define them in each group at the launch file, but it didn't succeed. How can I remap the follow_joint_trajectory to the two different namespaces?
2. After it will launch the two robots, am I suppose to see the two arms on rviz? It seems that I need to launch the moveit_planning execution at different namespace. Am I right?

My goal is to launch two arms in the simulator where I can check the collaboration in the task. I plan to create node the will sync the task command between them.
I can create two packages, but I want to use the same code, just send different commands.

gavanderhoorn, 2018-06-14:
re 1: you need to make sure you specify the correct controller namespace in your `controller.yaml` (or whatever name you've used) in your moveit `config` folder. No remapping should be needed.

re 2: not by default. You'll need to create a new xacro macro and top-level xacro that combines the two arms in a single scene. That is, if you want both arms / MoveIt to be aware of each other / it.

Multi-group setups with MoveIt are easiest if you put everything in a single, combined scene, but keeping everything separate (with namespaces) is definitely possible.

gavanderhoorn, 2018-06-14:
As this seems to no longer be an issue with the packages in this repository I'm going to close this.

Feel free to keep commenting on it, but do note that I believe other support channels would be more appropriate for your follow-up questions.

closed 2018-06-14 by gavanderhoorn