ros-industrial/kuka_experimental #136: Can KRL be run in ROS simulation?

hanwin-tbd, 2018-07-02:
I can launch KUKA KR6R700 robot simulation in both RViz and web (using ROS3DJS). I can also control it by using joint state publisher in both. Now, I would like to run KRL to move that robot in simulation. Is it possible?

gavanderhoorn, 2018-07-02:
No, not with the packages in this repository.

That would require a KRL interpreter and execution engine, and we don't have that / supply that.

gavanderhoorn, 2018-07-02:
But I must also point out that what you ask sounds like *replacing* the KUKA controller, and that is specifically not one of the goals of ROS-Industrial and / or the packages in this repository.

gavanderhoorn, 2018-07-02:
As I got curious, I looked for some literature on this subject.

After spending 2 mins with Google, I found [On reverse-engineering the KUKA Robot Language](https://arxiv.org/pdf/1009.5004.pdf) as the only relevant reference. The authors appear to have done something like create a KRL interpreter, but it's implemented on top of their own infrastructure (Robotics API), which does not seem to be available (but I haven't searched for it).

gavanderhoorn, 2018-07-02:
As your question is interesting @hanwin-tbd, but is not an issue with the pkgs in this repository, nor is immediately actionable, I'm going to close it.

If you want to continue discussing, please feel free to keep commenting.

closed 2018-07-02 by gavanderhoorn