ros-industrial/kuka_experimental #147: "Jerky" motions while using kuka_rsi_hw_interface for a KUKA KR240 L210 MED

BastiSta, 2018-12-05:
Hi all,
I am working with a KUKA KR240 L210 MED with a safe control system and would like to control it via the kuka_rsi_hw_interface. Starting the moveit_planning_execution.launch works and the robot can be started without problems. Also the visualization and path planning via Rviz and the execution of the movement works. My problem now is that the robot jerks randomly and does not move smoothly. At first I thought, there are too few waypoints passed and increased the number but there was no improvement. Also different planners from moveit don't lead to a clear improvement of the movement.
Has someone already had similar problems and could help me?

cschindlbeck, 2018-12-05:
Yes. This is the issue discussed in https://github.com/ros-industrial/kuka_experimental/issues/126 and unfortunately it has not been resolved yet. I slow down the robot to mitigate this issue at least a little bit.

BrettHemes, 2018-12-05:
I have the same issues with the EKI interface as well... @cschindlbeck @gavanderhoorn are you running with this/these issues currently?

gavanderhoorn, 2019-07-18:
There are two main causes for jerky motions:

 1. OMPL planning in combination standard IPTP
 1. non-jerk limited/aware time parameterisation on the MoveIt side (and the `joint_trajectory_controller`)

Switching to the `AddIterativeSplineParameterization` or `AddTimeOptimalParameterization` helps, but won't completely fix everything.

Until somebody writes a jerk-limited time parameterisation and `joint_trajectory_controller` *or* implements some low-pass filtering on the controller side I believe we're going to keep 'suffering' from this.