ros-industrial/kuka_experimental #149: ROS cannot connect KRC2 with RSI

zhihaoros, 2019-02-03:
Hello everyone,

Unfortunately I cannot connect KRC2 with RSI 2.3 to ROS after using your code and following the README.md very carefully. After I start roslaunch kuka_rsi_hw_interface test_hardware_interface.launch sim:=false, I get the following output: 

```
PARAMETERS                  
 * /controller_joint_names: ['joint_1', 'join...
 * /joint_state_controller/publish_rate: 50
 * /joint_state_controller/type: joint_state_contr...
 * /position_trajectory_controller/action_monitor_rate: 20
 * /position_trajectory_controller/joints: ['joint_1', 'join...
 * /position_trajectory_controller/state_publish_rate: 50
 * /position_trajectory_controller/type: position_controll...
 * /robot_description: <?xml version="1....
 * /rosdistro: kinetic
 * /rosversion: 1.12.14
 * /rsi/listen_address: 192.168.1.2
 * /rsi/listen_port: 6008

NODES
  /
    controller_spawner (controller_manager/spawner)
    kuka_hardware_interface (kuka_rsi_hw_interface/kuka_hardware_interface_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

ROS_MASTER_URI=http://localhost:11311

process[kuka_hardware_interface-1]: started with pid [5504]
[ INFO] [1548845612.858475019]: Starting hardware interface...
process[controller_spawner-2]: started with pid [5505]
[ INFO] [1548845612.866874200]: Loaded kuka_rsi_hardware_interface
[ INFO] [1548845612.871368981]: Setting up RSI server on: (192.168.1.2, 6008)
process[robot_state_publisher-3]: started with pid [5518]
[ INFO] [1548845612.933509003]: Waiting for robot!
[WARN] [1548845613.354319]: DEPRECATION warning: --shutdown-timeout has no effect.
[INFO] [1548845613.354845]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1548845613.357367]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1548845613.359516]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1548845613.362069]: Loading controller: position_trajectory_controller
[INFO] [1548845613.414381]: Loading controller: joint_state_controller
[INFO] [1548845613.425263]: Controller Spawner: Loaded controllers: position_trajectory_controller, joint_state_controller
```

and I start the ros_rsi.src only to get the error:

`SEN: RSI execution error <execute> -RSI stopped`

and in the rsiErrors.log I see the following error:

`Error: ERX: Defined timeout limit reached. Answer from server missed!`

I have been stumped at here for many days. Could anyone give me some suggestions how to figure this out?
@gavanderhoorn @ivareri @destogl @ccopot @ipa-nhg

ivareri, 2019-02-05:
I no longer have easy access to the lab, but if I remember correctly the error boils down to the KRC not being able to connect to the ROS pc. 

Some basic questions to help narrow down the scope:
1. Have you had RSI ethernet working on this controller before?  
2. How is the network cabled? (through a switch, are both network cards on the KRC connected, etc)
3. What is the IP for the RSI card?
4. What is the KRC windows IP?

zhihaoros, 2019-02-06:
@ivareri  Thanks for your reply.

>     1. Have you had RSI ethernet working on this controller before?

There was someone who had also used this krc2 with RSI in 2015, because I can see the errors in 2015 in the rsiErrors.log. It seems that he also failed to connect krc2 to ROS. His error was: 

`can't connect to ip 192.168.1.1, errors while opening the channel!`

I think the reason to this error is that he used tcp/ip protocol in the .xml file.

>     2. How is the network cabled? (through a switch, are both network cards on the KRC connected, etc)

network card 1: Intel(R) PRO/1000 CT Network Connection (192.168.1.5, 255.255.255.0)
network card 2: Realtime OS virtual Network (192.0.1.2, 255.255.255.0)
An ethernet network cable is used between PC and KRC2.

>     3. What is the IP for the RSI card?

192.168.1.5 (it was set up while installing RSI 2.3)

>     4. What is the KRC windows IP?

192.168.1.5

gavanderhoorn, 2019-02-15:
(late response) I'm not an expert here, but can you use any of the RSI example programs? That would tell us whether it's a RSI setup problem or something wrong with the RSI driver in this repository.

zhihaoros, 2019-02-15:
@ivareri @gavanderhoorn  Thanks for your reply.

The successful connection has just appeared in terminal:

```
NODES
  /
    controller_spawner (controller_manager/spawner)
    kuka_hardware_interface (kuka_rsi_hw_interface/kuka_hardware_interface_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)

auto-starting new master
process[master]: started with pid [5506]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 3a0a3f28-311f-11e9-9ece-180373dcb049
process[rosout-1]: started with pid [5520]
started core service [/rosout]
process[kuka_hardware_interface-2]: started with pid [5523]
process[controller_spawner-3]: started with pid [5538]
[ INFO] [1550234581.430348354]: Starting hardware interface...
process[robot_state_publisher-4]: started with pid [5539]
[ INFO] [1550234581.437757616]: Loaded kuka_rsi_hardware_interface
[ INFO] [1550234581.439154087]: Setting up RSI server on: (192.168.0.1, 6008)
[ INFO] [1550234581.484863546]: Waiting for robot!
[INFO] [1550234581.910961]: Controller Spawner: Waiting for service controller_manager/load_controller
[INFO] [1550234581.913237]: Controller Spawner: Waiting for service controller_manager/switch_controller
[INFO] [1550234581.915310]: Controller Spawner: Waiting for service controller_manager/unload_controller
[INFO] [1550234581.917478]: Loading controller: position_trajectory_controller
[INFO] [1550234581.971071]: Loading controller: joint_state_controller
[INFO] [1550234581.978656]: Controller Spawner: Loaded controllers: position_trajectory_controller, joint_state_controller
[ INFO] [1550234709.048009054]: Got connection from robot
[INFO] [1550234709.060804]: Started controllers: position_trajectory_controller, joint_state_controller
```
However, after I run this code according to the README.md:

`rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller`

I have changed the parameter of joint in this GUI controller, but my robot cannot move. My question is, can the robot actually move by using this controller? Can RSI actually know the ROS message published by Robot state publisher?

gavanderhoorn, 2019-02-16:
> @ivareri @gavanderhoorn Thanks for your reply.
> 
> I only used the RSI driver in this repository. I used the KRL demo program of the CD because the KRL in this repository cannot be compiled by my KRC Software.

Which version do you have? What were the errors?

The [KR C2 version](https://github.com/ros-industrial/kuka_experimental/tree/0f03b98238ade2389770cb148aca724e4fd1580c/kuka_rsi_hw_interface/krl/KR_C2) of the KRL programs (did you use that version?) is supposed to be compatible with RSI 2.3, so it would be good to know what didn't work, as we can perhaps update / fix things.

> The successful connection has just appeared in terminal:
> 
> ```
> NODES
>   /
>     controller_spawner (controller_manager/spawner)
>     kuka_hardware_interface (kuka_rsi_hw_interface/kuka_hardware_interface_node)
>     robot_state_publisher (robot_state_publisher/robot_state_publisher)
> 
> auto-starting new master
> process[master]: started with pid [5506]
> ROS_MASTER_URI=http://localhost:11311
> 
> setting /run_id to 3a0a3f28-311f-11e9-9ece-180373dcb049
> process[rosout-1]: started with pid [5520]
> started core service [/rosout]
> process[kuka_hardware_interface-2]: started with pid [5523]
> process[controller_spawner-3]: started with pid [5538]
> [ INFO] [1550234581.430348354]: Starting hardware interface...
> process[robot_state_publisher-4]: started with pid [5539]
> [ INFO] [1550234581.437757616]: Loaded kuka_rsi_hardware_interface
> [ INFO] [1550234581.439154087]: Setting up RSI server on: (192.168.0.1, 6008)
> [ INFO] [1550234581.484863546]: Waiting for robot!
> [INFO] [1550234581.910961]: Controller Spawner: Waiting for service controller_manager/load_controller
> [INFO] [1550234581.913237]: Controller Spawner: Waiting for service controller_manager/switch_controller
> [INFO] [1550234581.915310]: Controller Spawner: Waiting for service controller_manager/unload_controller
> [INFO] [1550234581.917478]: Loading controller: position_trajectory_controller
> [INFO] [1550234581.971071]: Loading controller: joint_state_controller
> [INFO] [1550234581.978656]: Controller Spawner: Loaded controllers: position_trajectory_controller, joint_state_controller
> [ INFO] [1550234709.048009054]: Got connection from robot
> [INFO] [1550234709.060804]: Started controllers: position_trajectory_controller, joint_state_controller
> ```
> 
> However, after I run this code according to the README.md:
> 
> `rosrun rqt_joint_trajectory_controller rqt_joint_trajectory_controller`
> 
> I have changed the parameter of joint in this GUI controller, but my robot cannot move. My question is, can the robot actually move by using this controller? Can RSI in KRC2 actually subscribe to the ROS topic published by Robot state publisher in PC?

I'm not sure what the "robot state publisher" has to do with this (at least, if you're talking about [this one](http://wiki.ros.org/robot_state_publisher)).

As to `rqt_joint_trajectory_controller`: yes, your robot should move, provided the required controller is active, you've pressed the big red enable button, RSI is still active on the robot side and there are no errors shown anywhere.

zhihaoros, 2019-02-16:
> > @ivareri @gavanderhoorn Thanks for your reply.
> > I only used the RSI driver in this repository. I used the KRL demo program of the CD because the KRL in this repository cannot be compiled by my KRC Software.
> 
> Which version do you have? What were the errors?
> 
> The [KR C2 version](https://github.com/ros-industrial/kuka_experimental/tree/0f03b98238ade2389770cb148aca724e4fd1580c/kuka_rsi_hw_interface/krl/KR_C2) of the KRL programs (did you use that version?) is supposed to be compatible with RSI 2.2 (probably 2.3 as well), so it would be good to know what didn't work, as we can perhaps update / fix things.

I use RSI 2.3.2 and KRC2. In your repository there is only .src program. However, for KRC2 there must be .dat program accompanied with .src program. Otherwise the KRC Software won't show the .src program. 

> I'm not sure what the "robot state publisher" has to do with this (at least, if you're talking about [this one](http://wiki.ros.org/robot_state_publisher)).
>
> As to `rqt_joint_trajectory_controller`: [...]

As you can see, there are altogether 3 Nodes in this RSI driver:

```
NODES
  /
    controller_spawner (controller_manager/spawner)
    kuka_hardware_interface (kuka_rsi_hw_interface/kuka_hardware_interface_node)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
```

In the GUI `rqt_joint_trajectory_controller`, all the initial joint data are the one which are in my URDF. So I guess what   `robot_state_publisher` has done is just publishing the data of URDF file and then  `rqt_joint_trajectory_controller` subscribes this data and shows them in GUI.

gavanderhoorn, 2019-02-17:
> I use RSI 2.3.2 and KRC2. In your repository there is only .src program. However, for KRC2 there must be .dat program accompanied with .src program. Otherwise the KRC Software won't show the .src program.

Hm. We haven't had any reports/complaints about this, so it would seem that other users have been able to work with the files that are there.

As I don't have access to KR C2, it would be very much appreciated if you could try to address what is missing and then submit a PR, so future users can avoid running into what you've had to figure out.

> As you can see, there are altogether 3 Nodes in this RSI driver:

To be precise: there are three nodes started by the launch file that you used.

The driver itself is really only just the `kuka_hardware_interface`.

The rest is part of `ros_control` (`controller_spawner`) and a standard ROS node that transform `JointState`s into TF frames (`robot_state_publisher`).

> ```
> NODES
>   /
>     controller_spawner (controller_manager/spawner)
>     kuka_hardware_interface (kuka_rsi_hw_interface/kuka_hardware_interface_node)
>     robot_state_publisher [...]

No, that's not how this works.

`rqt_joint_trajectory_controller` is a component of `ros_control`. It communicates directy with the `JointTrajectoryController` configured to be loaded and started (notice the `position_trajectory_controller` in the roslaunch output that you included in your previous comments).

> all the initial joint data are the one which are in my URDF

Which urdf are you using? The zero-position is very important, and not necessarily very obvious with KUKAs.

zhihaoros, 2019-02-17:
> Hm. We haven't had any reports/complaints about this, so it would seem that other users have been able to work with the files that are there.
>
> As I don't have access to KR C2, it would be very much [...]

I think I gotta contact @ivareri who has ever tested it with KRC2 ed05. My KRC2 version is KRC2 sr. Maybe these are two different versions of KRC2 and have different mechanisms.

> Which urdf are you using? The zero-position is very important, and not necessarily very obvious with KUKAs.

I use the URDF which was created by the one who has ever used this KR5 sixx R850 with KRC2.

ivareri, 2019-02-22:
Sorry for the late reply, I try to respond when i can, but most of my attention and time is required elsewhere these days. 

> I think I gotta contact @ivareri who has ever tested it with KRC2 ed05. My KRC2 version is KRC2 sr. Maybe these are two different versions of KRC2 and have different mechanisms.

The .DAT file is just for storing data, AFAIK it is not required for programs to run. If the SRC file isn't copied into the work directory using KSS it will not show up until you restart KSS. Could this be the problem?  You could try create a new, blank, program, and then copy the code into the SRC file.  

What version of KSS are you using?

zhihaoros, 2019-02-23:
> The .DAT file is just for storing data, AFAIK it is not required for programs to run. If the SRC file isn't copied into the work directory using KSS it will not show up until you restart KSS. Could [...]

I use KSS 7.0.14. The process that I copy the .src file of this repository and paste it is that: I create a new file in the KSS and then minimize the KSS, open the C:\KRC\program\R1, then open the file that I have just created and paste the .src file of this repository. After a coldstart then the new file that i have just created disappears.

ivareri, 2019-02-24:
> I use KSS 7.0.14. 

IIRC I used KSS 5.5. I don't know what has changed from 5.x to 7.x, so it might be a required to use DAT files in 7.x. 

> The process that I copy the .src file of this repository and paste it is that: I create a new file in the KSS and then minimize the KSS, open the C:\KRC\program\R1, then open the file that I have [...]

Are you able able to paste the content of the file using KSS? (Open the file from the repo in notepad, copy text and paste into editor i KSS.) That way you should get warnings if there are any errors in the code.

zhihaoros, 2019-02-25:
> Are you able able to paste the content of the file using KSS? (Open the file from the repo in notepad, copy text and paste into editor i KSS.) That way you should get warnings if there are any errors in the code.

Unfortunately I cannot paste anything into editor.

bootsch, 2019-02-25:
In my workflow I copy .src files to somewhere on C: using a shared folder.
from there I copy the file into R1/programs using copy/paste on the file.

If you copy straight to R1 the file will dissapear after reboot as you noticed.

zhihaoros, 2019-02-25:
> In my workflow I copy .src files to somewhere on C: using a shared folder. from there I copy the file into R1/programs using copy/paste on the file.

I‘m sorry i cannot understand where is "somewhere" and what is "shared folder".

bootsch, 2019-02-25:
To be more explicit

I connect to the KRC2 over ethernet.
On our KRC2 we setup Windows shared folders and have a directory c:\upload
We mount the shared disk on our pc and drop the .src files in the folder.

Then on the pendant we log in as administrator and browes to c:\upload
select the file you want to use and copy the file using right-click copy.
Go to R1/Program and use the menu to paste the file.

> 
> In my workflow I copy .src files to somewhere on C: using a shared folder. from there I copy the file into R1/programs using copy/paste on the file.
> 
> I‘m sorry i cannot understand where is "somewhere" and what is "shared folder".

Paul Boots
bootsch@acm.org <mailto:bootsch@acm.org>

gavanderhoorn, 2019-07-18:
Closing this due to inactivity.

closed 2019-07-18 by gavanderhoorn