ros-industrial/kuka_experimental #153: RSI 4.0 - missing .src

xenojimmy, 2019-04-03:
regarding the latest update for 4.0+ 
the ros_rsi.src file is not included.

I'm asking if the ros_rsi.src file changes from the previous versions just by removing the .rsi extension in the first argument of RSI_CRATE( "ros_rsi" ) instead of  RSI_CRATE( "ros_rsi.rsi" ), as it used to be in older versions.

I get RSI_CREATE: invalid index - signal output error when launching it

thank you

gavanderhoorn, 2019-04-03:
With "latest update" you are referring to #152?

If so, that is still under review, so please comment on that PR instead of opening an issue.

gavanderhoorn, 2019-04-03:
I'm going to assume you're actually asking about #152, so I'm closing this.

Let's continue in the PR.

closed 2019-04-03 by gavanderhoorn