ros-industrial/kuka_experimental #156: Missing links' inertias and masses (LBR iiwa)

ferrolho, 2019-07-18:
The KUKA LBR iiwa 14 model is missing all the links' inertia and mass.
More, the joints effort is set to `0`.

For example, check the code snippet below:
https://github.com/ros-industrial/kuka_experimental/blob/0f03b98238ade2389770cb148aca724e4fd1580c/kuka_lbr_iiwa_support/urdf/lbr_iiwa_14_r820_macro.xacro#L33-L46

gavanderhoorn, 2019-07-18:
Yes, you're correct. This is known.

Did you expect something different?

ferrolho, 2019-07-18:
Yes, I'd expect the model to contain those data. If it is known they are missing, is there any reason why it has not been addressed? Is it just that no one raised the issue before, or is there some ulterior motive?

gavanderhoorn, 2019-07-18:
> is there some ulterior motive?

that sounds rather nefarious.

> Yes, I'd expect the model to contain those data.

I believe you are assuming that all models of LBR iiwa are automatically created for use in simulations. That is not the case. And especially not for this model.

All usage of this model so far has not had any use for `inertial` tags or effort limits.

If you have use for those, and would like to use this model, then please add them and submit a PR.

Note btw that it's rare for industrial robots to have information available about *actual* inertias, masses and effort limits.

ferrolho, 2019-07-18:
> that sounds rather nefarious.

Sorry, my bad; I've had the wrong meaning of 'ulterior' in mind all along and didn't know it had a negative connotation. What I meant to ask was if there is a reason why they are purposefully omitted, e.g., ease of maintenance, shorter/uncluttered `.urdf`.

> I believe you are assuming that all models of LBR iiwa are automatically created for use in simulations. That is not the case. And especially not for this model.

Got it.

> If you have use for those, and would like to use this model, then please add them and submit a PR.

Yes, I'd like to use this model not only for visualisation but also to extract its dynamic properties for use in dynamics optimisation. I will happily submit a PR if I can find the intertias, masses and efforts. Any pointers to specs sheets or other resources would be appreciated!

ferrolho, 2020-08-13:
I'd like to bring this to a closure by referring to [IFL-CAMP/iiwa_stack](https://github.com/IFL-CAMP/iiwa_stack), which provides good ROS support for LBR iiwa 7 and 14. It also contains the inertia parameters, e.g., [IFL-CAMP/iiwa_stack/blob/master/iiwa_description/urdf/iiwa14.xacro#L74-L79](https://github.com/IFL-CAMP/iiwa_stack/blob/a4f132ad093d276e63040ddbbac6b2cd1facbdcc/iiwa_description/urdf/iiwa14.xacro#L74-L79).

closed 2020-08-13 by ferrolho

gavanderhoorn, 2020-08-13:
Great. Thanks for reporting back.

> It also contains the inertia parameters

Which also look like approximations, guestimates or the result of system identification.

I'm not aware of any publicly available resources which contain the official (as far as those exist) dynamic parameters of IIWAs.