ros-industrial/kuka_experimental #157: > From my previous experience with ROS-I driver it was always stream trajectory first and then execute (ABB, Motoman for example)

VastDong, 2019-07-25:
> From my previous experience with ROS-I driver it was always stream trajectory first and then execute (ABB, Motoman for example)

`abb_driver` implements a *download* approach, `motoman_driver` is a *streaming* driver. The former uploads (download from the controller's pov) the entire trajectory, then executes. The latter streams a trajectory pointwise (with a small buffer on the controller side).

> I thought that the role of ROS-I driver is to allow performing plan&execute approach.

Yes? Both downloading and streaming would seem to support that.

> Why real-time control in case of KUKA?

Because it covers both the use-cases that more 'traditional' ros-i drivers support, as well as more high-fidelity (high-bw) and real-time use-cases (such as visual servoing). We use `ros_control` (and specifically `joint_trajectory_controller`) to in the end expose a ROS-I compatible `JointTrajectory` action interface, which hides all the differences (while enjoying all the advantages of a higher-bw interface).

Btw: there is no *requirement* for ROS-I drivers to use the common infrastructure and messaging system provided by the `industrial_robot_client` packages: `simple_message` is by no means a panacea, and especially for vendors that already provide good external motion interfaces (ABB, KUKA, Mitsubishi, etc) those are preferred, as they typically allow for much better trajectory execution and motion control.

_Originally posted by @gavanderhoorn in https://github.com/ros-industrial/kuka_experimental/issues/77#issuecomment-279993753_

VastDong, 2019-07-25:
Or any one have the experiment of implementing visual servoing with KUKA RSI. Could you please help me ?

BrettHemes, 2019-10-09:
You need to ensure that the trajectory data streamed to the robot via RSI is sufficiently smooth.  Despite only taking position data (at 4ms intervals) the positions should be continuous and differentiable.  From experience, C2 continuity (via trapezoidal motion profiles) is sufficient (i.e., continuous velocity / bounded acceleration)

gavanderhoorn, 2021-03-30:
I'm going to close this issue due to inactivity.

Implementing a Visual Servoing loop over RSI is certainly possible, but it's on the user to guarantee smoothness of the trajectories.

Whether that's achieved on the external side (ie: PC) or using one of RSIs filtering components depends on perhaps non-functional requirements.

closed 2021-03-30 by gavanderhoorn