ros-industrial/kuka_experimental #160: Failed to read from robot EKI server

TomasMerva, 2019-10-09:
Hello, I am trying to connect ROS melodic with KUKA robot `KR6 r900 sixx` using kuka-experimental. However, after having done all things in README.md about configuring EKI on the controller, I am getting the error:

`Failed to read from robot EKI server within alloted time of 5 seconds. Make sure eki_hw_interface is running on the robot controller and all configurations are correct.`

At the side of KUKA controller I have done these things:
    1. setting IP, creating new port 54600 (for udp)
    2. copy `EkiHwInterface` into `C:\KRC\ROBOTER\Config\User\Common\EthernetKRL\` (with same  IP as in the previous step)
    3. copy kuka_eki_hw_interface.dat and kuka_eki_hw_interface.src to `KRC:\R1\Program`

At the side of ROS I have done:
1. The hardware_controllers.yaml and controller_joint_names.yaml I have not changed so the names of joints are as follows:

        #Publish all joint states
        joint_state_controller:
          type: joint_state_controller/JointStateController
          publish_rate: 50

        # Joint trajectory controller
        position_trajectory_controller:
          type: "position_controllers/JointTrajectoryController"
          joints:
            - joint_a1
            - joint_a2
            - joint_a3
            - joint_a4
            - joint_a5
            - joint_a6

          state_publish_rate: 50 # Defaults to 50
          action_monitor_rate: 20 # Defaults to 20

    Is the fact that I have written here the wrong names of the joints why I cant read from EKI server?

The test_params.yaml file is as follows:

    eki:
      robot_address: "192.168.100.2"
      robot_port: "54600"
      socket_timeout: 5

My launch file is as follows:

    <?xml version="1.0" encoding="utf-8"?>
    <launch>
        <param name="robot_description" command="$(find xacro)/xacro.py '$(find kuka_kr6_support)/urdf/kr6r900sixx.xacro'"/>

       <rosparam file="$(find kuka_eki_hw_interface)/test/test_params.yaml" command="load" />

       
       <node name="kuka_eki_hardware_interface" pkg="kuka_eki_hw_interface"
         type="kuka_eki_hw_interface_node"
         respawn="false"
         output="screen"
         required="true"/>

       
       <rosparam file="$(find kuka_eki_hw_interface)/config/hardware_controllers.yaml"
         command="load"/>

       
       <rosparam file="$(find kuka_eki_hw_interface)/config/controller_joint_names.yaml"
         command="load"/>

       
       <node name="controller_spawner" pkg="controller_manager" type="spawner" respawn="false"
         output="screen" args="position_trajectory_controller joint_state_controller --shutdown-timeout 1"/>
       
       <node name="robot_state_publisher" pkg="robot_state_publisher" type="robot_state_publisher" />
    </launch>

Does anyone see where the mistake might be?

Thank you all in advance

TomasMerva, 2019-10-09:
https://answers.ros.org/question/334974/failed-to-read-from-robot-eki-server/

gavanderhoorn, 2019-10-09:
Seeing as we already have the post on ROS Answers, I'm going to close this one.

closed 2019-10-09 by gavanderhoorn