ros-industrial/kuka_experimental #163: Connection to KR210R2700

storeyp, 2020-01-14:
Hi,
I'm experimenting with connection to our Kuka KR210, but being new to ROS I might be hitting some basic problems while trying to connect.

At the moment, I'm getting the following when trying to do a test_hardware_interface.launch:

```
[ERROR] [1579000815.735010136]: Could not load controller 'position_trajectory_controller' because controller type 'position_controllers/JointTrajectoryController' does not exist.
[ERROR] [1579000815.735085694]: Use 'rosservice call controller_manager/list_controller_types' to get the available types
[ERROR] [1579000816.736960]: Failed to load position_trajectory_controller
```

The rosservice call returns the following
```
types: [diff_drive_controller/DiffDriveController, joint_state_controller/JointStateController,
  position_controllers/JointGroupPositionController, position_controllers/JointPositionController]
base_classes: ['controller_interface::ControllerBase', 'controller_interface::ControllerBase', 'controller_interface::ControllerBase',
  'controller_interface::ControllerBase']
```

I have tried fixing it via a shotgun approach of:
```
git clone https://github.com/ros-controls/ros_controllers.git
rosdep install --from-paths src --ignore-src -r -y
catkin_make
```

But I still get the same errors.

Any help or direction for troubleshooting appreciated!

closed 2020-01-14 by storeyp

gavanderhoorn, 2020-01-14:
Which installation instructions did you follow? Have you followed the regular ROS tutorials already? It will not immediately solve your problem, but it may make it somewhat easier for you to understand what is going on.

Don't install anything from source if you don't have to.

Remove `ros_controllers` from your `src` workspace and rebuild your workspace (probably have to remove your `build` and `devel` folders first).

Now follow the procedure in the answer of #q252478 (be sure to use the correct ROS version identifier of course).

storeyp, 2020-01-14:
I realised my mistake just as you replied. It would help if I installed it into `src` wouldn't it!
That little bug seems to be fixed now, although I don't know if installing all of `ros_controllers` is really best practice like you say. I'll go through the link you sent.

Regarding tutorials, yep I've been through some training, and I've signed up to the edX course starting this week, so hopefully I can begin asking some harder-hitting questions in future!

Thanks for the quick response!