ros-industrial/kuka_experimental #170: kr10r1420

dbodily1, 2020-08-06:
Are there any plans to release more kuka models?  Specifically I'm looking for KR10-R1420

gavanderhoorn, 2020-08-06:
There are no plans to add more models.

Note: this repository is labelled as [image], which means contributions will have to come from community members.

There is no roadmap driven by the consortia.

gavanderhoorn, 2020-08-06:
As this is not an issue with the packages in this repository, I'm going to close it.

Feel free to keep commenting of course.

closed 2020-08-06 by gavanderhoorn