ros-industrial/kuka_experimental #171: Support for Kuka robot.

ranjitkathiriya, 2020-11-04:
I have ordered KR20 R1810-2. Can I please get the urdf file for that robot? And I also want the support for that file in ROS. If possible, can you help me with this?

I have already talked with the Kuka team that has told me to contact this repository.

Thanks,
Ranjit

gavanderhoorn, 2020-11-04:
We're more than willing to help you, but this is not a place where you can "order" URDFs or ROS support for robots.

I would suggest you take a look at some of the existing robot support packages and create one for the robot you mention. I believe most work would be in converting the meshes to a format we can use and to configure the kinematic and dynamics properties.

gavanderhoorn, 2020-11-04:
As this is not an issue with the packages in this repository, I'm going to close it.

Feel free to keep commenting on the issue of course.

closed 2020-11-04 by gavanderhoorn

ranjitkathiriya, 2020-11-11:
I have ordered KR20 R1810-2, does it is supported by ROS? If not then I have to cancel the robot. I am so confused

gavanderhoorn, 2020-11-11:
First, and this will be pedantic, but I believe also important:

> is supported by ROS?

"ROS" does not support anything. *Only* packages will provide support for specific pieces of hardware.

Reason I feel this is important is because this is a mistake new users often make, and limits their use of ROS significantly. They make a binary decision about whether or not to use ROS based on the presence or absence of one particular feature, and consider it as a monolithic whole, instead of the granular, component based system it is.

With that out of the way: your question appears to be:

> are there packages available which allow me to use my KR20 R1810-2 *with* ROS?

If this is your question, then the answer consists of two parts:

 1. if you have ordered either RSI or EKI with your robot controller, then you have the necessary interfaces on your controller to allow either `kuka_eki_hw_interface` or `kuka_rsi_hw_interface` to control it.

    If you don't have RSI, nor EKI, it doesn't matter which robot model you have, as you will not be able to control it using any of the packages we host here in this repository (if you don't have RSI nor EKI, you could still use something like KUKAVARPROXY, but that's not supported here, so I won't consider those).

 1. the second part of the answer would be: there is no *robot support package* right now for the specific robot model you mention. Support packages contain configuration, `.launch` and `.xacro` files which describe the kinematics and physical shape of your robot. Without those, you could still use `kuka_eki_hw_interface` or `kuka_rsi_hw_interface`, but you would not be able to visualise anything in RViz, nor would MoveIt be able to plan for your robot, as it would not have the required information.

Summarising: *yes*, you could use your robot with `kuka_eki_hw_interface` or `kuka_rsi_hw_interface` provided you have either RSI or EKI on your controller. *no* you cannot use any of the existing robot support packages in this repository, as those do not support your specific robot model and variant.

Finally: `kuka_eki_hw_interface` or `kuka_rsi_hw_interface` are robot *agnostic*. There are no assumptions about which particular robot model they are being used with. In fact, they don't even know anything about robots. They only pack and unpack RSI and EKI packets. The fact that somewhere, something is using that to control a KUKA robot is completely unimportant.

gavanderhoorn, 2020-11-11:
So because of this:
> 2\. there is no _robot support package_ right now for the specific robot model you mention.

you would have to do some work, which essentially comes down to creating a support package for your KR20 R1810-2.

That is not too difficult, and is probably easiest by looking at existing robot support packages in this repository.

In a nutshell you'd take let's say the KR16's package, copy it, then update all references to KR16 to read KR20. Then update `link` lengths, `joint` limits and meshes. Check `.launch` files and configuration files.

That would be about it.

At that point you'd have a basic robot support package for the KR20 series, with one variant supported, namely the `R1810-2`.

ranjitkathiriya, 2021-04-01:
Hello @gavanderhoorn,

I have checked that there is a huge difference between KR16 and KR20 R1810-2 links. Can we get .dae files and .stl files from Kuka robotic if we ask them? I think it will take a high time if we create the robot into blender or Solidworks. Any suggestion on this issue?
 
[image]
[image]

gavanderhoorn, 2021-04-01:
> I have checked that there is a huge difference between KR16 and KR20 R1810-2 links.

I'm a bit surprised you are surprised about this.

I would not have expected anything else.

Why are you using the KR16 support package with a different robot?

> Can we get .dae files and .stl files from Kuka robotic if we ask them?

Have you checked the [download centre](https://www.kuka.com/en-gb/services/downloads?terms=Language:en:1&q=)?

This is all standard KUKA robot usage.

They are not going to provide anything specifically geared towards ROS usage.

ranjitkathiriya, 2021-04-01:
Thanks for replying to my answers @gavanderhoorn,

> Why are you using the KR16 support package with a different robot?

There are two reasons. First, I was not getting any package for my robot, and I also tried to create a package, but I think with 3d software, it is time taking to create one. Second, I wanted to check that I can communicate with RSI and further can able to integrate and jog with the moveit. I have faced some controller version issues, but finally, I can jog via move it.

While jogging with move it, I have seen around a 7-8 cm difference when jogging with the real-world and virtual world. 

> Have you checked the download center?

Yes, I have checked that already. I am not getting any Solidworks or any 3d model files for this robot.

Can I ask a question?

How did you get those at the time of generation of this package? Did you make this on your own? If yes, then which software have you used for that? Or you got these files from Kuka?

gavanderhoorn, 2021-04-01:
Everything comes from KUKA. Either directly, or via the download centre.

gavanderhoorn, 2021-04-01:
> Yes, I have checked that already. I am not getting any Solidworks or any 3d model files for this robot.

So [these](https://www.kuka.com/en-gb/services/downloads?terms=Language:en:1;Category:CAD;&q=1810) are results applicable to your robot?

ranjitkathiriya, 2021-04-01:
> So these are not results applicable to your robot?

Thanks! with this link I got a Parasolid file, but how can I convert this file to .dae and .stl (The same file format that is used by urdf or macro tag for mesh)

gavanderhoorn, 2021-04-01:
> but how can I convert this file to .dae and .stl (The same file format that is used by urdf or macro tag for mesh) for every link.

with a tool which can open parasolids and export them to other file formats.

This is a bit of work you'll have to do yourself, as there are no one-click approaches.

You *could* take a look at the Solidworks-to-URDF converter, but that of course requires Solidworks.

Personally I never use it, but to explain my own workflow would take more time than I have right now.

ranjitkathiriya, 2021-04-02:
Thank you so much for the help!

If I have any doubt, can I ask on this thread?

gavanderhoorn, 2021-04-02:
I would suggest we keep the issue tracker for actual issues with the code.

Questions about usage or how to approach certain tasks are best posted to ROS Answers.