ros-industrial/kuka_experimental #172: Using kuka_rsi_hw_interface: where to start?

wrd90, 2020-11-04:
Hello and sorry in advance for this noob question!

I have a `KUKA KRC4 Compact` controller and a `KR 10 R900 sixx` manipulator for which I would like to implement a path-correction application. As far as I understand it's possible to compute the path correction on a separate computer which is connected to the `KRC4 Compact` via `RSI` which uses the `EtherCat` interface (is that correct?). Now my problem is that I can't really find any documentation on how to do this. Could someone please point me to any helpful resources?

I would be entirely grateful for any hints on how to use the `kuka_rsi_hw_interface` with a `KRC4 Compact`

Best,
David

tingelst, 2020-11-04:
Hi David!

the `kuka_rsi_hw_interface` works with and was developed on the KRC4 Compact. You can find the installation procedure [here](https://github.com/ros-industrial/kuka_experimental/tree/indigo-devel/kuka_rsi_hw_interface/krl/KR_C4).

gavanderhoorn, 2020-11-04:
>  I would like to implement a path-correction application

While technically this would be possible with `kuka_rsi_hw_interface`, the assumption in that component is mostly that you'll be using RSI to "take over" control of a KUKA robot from a ROS PC. Overlaying corrections would be similar, but different from what the component currently does.

> which uses the `EtherCat` interface

Not completely. RSI is used over a regular ether*net* connection, which carries UDP datagrams.

It will be important to understand whether you're looking to implement ROS control over a KUKA, or whether you're going to be using RSI to overlay corrections. The former can be done after following the documentation linked to by @tingelst. The latter will require more work (and will require you to implement some parts yourself).

gavanderhoorn, 2020-11-04:
As this appears to be a usage/configuration question, and not so much a report about a problem with the packages in this repository, I'm going to close the issue.

If it turns out there is something to fix, we can re-open.

Feel free to keep commenting on it of course.

closed 2020-11-04 by gavanderhoorn

wrd90, 2020-11-04:
Hi to both of you, and thank you very much for your quick and helpful responses!

Ultimately my task is to implement a vision-based path correction. However, as it is yet in a very early stage I believe I could use the existing functionality to implement a simple movement and then overlay the correction, or also extend the functionality  as you suggested.  

First of all I will look into the readme that @tingelst provided (thank you again). In the meantime, @gavanderhoorn,  could you point me to where I would start digging in order to add a path-correction approach?

gavanderhoorn, 2020-11-11:
If you're just starting out, I would not use `kuka_rsi_hw_interface` as a template or starting point.

There are some assumptions in that hardware interface which you are immediately violating by not using RSI to completely control a KUKA.

While technically RSI *always* overlays offsets on motions, none of the `ros_control` controllers will be able to deal with this if you start using it as such. You *could* write a `ros_control` controller which can do this though. You might not even have to change anything in the hardware interface itself.