ros-industrial/kuka_experimental #173: Kuka_16 Urdf file while importing warnings

ranjitkathiriya, 2020-11-10:
While running the Kuka16 urdf file, I am getting this warning, and I am a bit confused that I should solve this and go further, or it is okay not to solve them?

But I am getting all details like axis and name of the robot.

Warnings:
Argv ['/home/ranjit/catkin_ws/src/moveit_kuka16_v2/scripts/robot_control_file.py']
[ INFO] [1605004485.172839739]: Loading robot model 'kuka_kr16_2'...
[ WARN] [1605004485.173736162]: Skipping virtual joint 'virtual_joint' because its child frame 'base' does not match the URDF frame 'base_link'
[ INFO] [1605004485.173775286]: No root/virtual joint specified in SRDF. Assuming fixed joint
[rospack] Error: package 'kuka_kr16_support' not found
[librospack]: error while executing command
[ERROR] [1605004485.259583343]: Error retrieving file [package://kuka_kr16_support/meshes/kr16_2/collision/base_link.stl]: Package [kuka_kr16_support] does not exist
[ WARN] [1605004485.259915289]: Link base_link has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[rospack] Error: package 'kuka_kr16_support' not found
[librospack]: error while executing command
[ERROR] [1605004485.299492297]: Error retrieving file [package://kuka_kr16_support/meshes/kr16_2/collision/link_1.stl]: Package [kuka_kr16_support] does not exist
[ WARN] [1605004485.300458185]: Link link_1 has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[rospack] Error: package 'kuka_kr16_support' not found
[librospack]: error while executing command
[ERROR] [1605004485.341026426]: Error retrieving file [package://kuka_kr16_support/meshes/kr16_2/collision/link_2.stl]: Package [kuka_kr16_support] does not exist
[ WARN] [1605004485.341248051]: Link link_2 has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[rospack] Error: package 'kuka_kr16_support' not found
[librospack]: error while executing command
[ERROR] [1605004485.382534189]: Error retrieving file [package://kuka_kr16_support/meshes/kr16_2/collision/link_3.stl]: Package [kuka_kr16_support] does not exist
[ WARN] [1605004485.382774149]: Link link_3 has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[rospack] Error: package 'kuka_kr16_support' not found
[librospack]: error while executing command
[ERROR] [1605004485.424590596]: Error retrieving file [package://kuka_kr16_support/meshes/kr16_2/collision/link_4.stl]: Package [kuka_kr16_support] does not exist
[ WARN] [1605004485.424884411]: Link link_4 has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[rospack] Error: package 'kuka_kr16_support' not found
[librospack]: error while executing command
[ERROR] [1605004485.464624243]: Error retrieving file [package://kuka_kr16_support/meshes/kr16_2/collision/link_5.stl]: Package [kuka_kr16_support] does not exist
[ WARN] [1605004485.464898400]: Link link_5 has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[rospack] Error: package 'kuka_kr16_support' not found
[librospack]: error while executing command
[ERROR] [1605004485.502623570]: Error retrieving file [package://kuka_kr16_support/meshes/kr16_2/collision/link_6.stl]: Package [kuka_kr16_support] does not exist
[ WARN] [1605004485.502847629]: Link link_6 has visual geometry but no collision geometry. Collision geometry will be left empty. Fix your URDF file by explicitly specifying collision geometry.
[ WARN] [1605004485.507866721]: No geometry is associated to any robot links
[ WARN] [1605004485.548761391]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
[ INFO] [1605004486.667352170]: Ready to take commands for planning group arm.
============ Planning frame: base_link
============ End effector link: link_6
============ Available Planning Groups: ['arm', 'gripping_tool']
============ Printing robot state
joint_state: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: "base_link"
  name: 
    - joint_a1
    - joint_a2
    - joint_a3
    - joint_a4
    - joint_a5
    - joint_a6
  position: [-1.341392802691654, -0.9516254018354844, -1.8231377433927103, 5.337112913441205, -0.2520891856416431, -3.2042488861662015]
  velocity: []
  effort: []
multi_dof_joint_state: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: "base_link"
  joint_names: []
  transforms: []
  twist: []
  wrench: []
attached_collision_objects: []
is_diff: False

Thanks,

gavanderhoorn, 2020-11-10:
>    `[rospack] Error: package 'kuka_kr16_support' not found`

This seems to be the root cause of all the other warnings and errors.

You'll want to make sure `rospack find kuka_kr16_support` returns the correct path, instead of "not found".

I just checked the `_macro.xacro` and everything seems to be in order, so for now I'm going to assume this is a configuration issue on your end.

Did you build the workspace, and then also `source /home/ranjit/catkin_ws/devel/setup.bash` before trying to run your script?

gavanderhoorn, 2020-11-10:
As this doesn't appear to be an issue with the package in this repository, I'm going to close it.

Feel free to keep commenting on it.

If it turns out this is something we need to address here, we'll re-open.

closed 2020-11-10 by gavanderhoorn

ranjitkathiriya, 2020-11-10:
Thanks for your reply Gavander,

source /home/ranjit/catkin_ws/devel/setup.bash - Yes I Always do this before running any file.

Can you please explain to me what steps you take to the simulation of the robot?

What I did was:
1. https://ros-planning.github.io/moveit_tutorials/doc/setup_assistant/setup_assistant_tutorial.html - I have followed these steps
2. I have given urdf file of kuka_16 (This robot is almost the same as Kuka_20  R1810-2 - the robot we have order)
3. While running the python script for this robot I am getting this warning. - https://ros-planning.github.io/moveit_tutorials/doc/move_group_python_interface/move_group_python_interface_tutorial.html

Is there any other way? Like I am new to this robot So please help me!! 

\Thanks
Ranjit

gavanderhoorn, 2020-11-11:
What is the output of `rospack find kuka_kr16_support` in your terminal(s)?

ranjitkathiriya, 2020-11-11:
The output for tha above question is 
rospack find kuka_kr16_support :  /home/ranjit/catkin_ws/src/kuka_experimental/kuka_kr16_support

and I have set up by doing git clone, and I am using urdf file of kuka_kr16_support while doing [MoveIt Setup Assistant](https://ros-planning.github.io/moveit_tutorials/doc/setup_assistant/setup_assistant_tutorial.html)