ros-industrial/kuka_experimental #175: Could not find a package configuration file provided by "roslaunch" with   any of the following names

ranjitkathiriya, 2020-11-30:
Hi there,

I am getting this error while building catkin_make. Can you please help me out? 

- +++ processing catkin metapackage: 'kuka_experimental'
-- ==> add_subdirectory(kuka_experimental/kuka_experimental)
-- +++ processing catkin package: 'kuka_kr10_support'
-- ==> add_subdirectory(kuka_experimental/kuka_kr10_support)
CMake Error at kuka_experimental/kuka_kr10_support/CMakeLists.txt:10 (find_package):
  By not providing "Findroslaunch.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "roslaunch", but CMake did not find one.

  Could not find a package configuration file provided by "roslaunch" with
  any of the following names:

    roslaunchConfig.cmake
    roslaunch-config.cmake

  Add the installation prefix of "roslaunch" to CMAKE_PREFIX_PATH or set
  "roslaunch_DIR" to a directory containing one of the above files.  If
  "roslaunch" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
See also "/home/kuka-ai/catkin_ws/build/CMakeFiles/CMakeOutput.log".
See also "/home/kuka-ai/catkin_ws/build/CMakeFiles/CMakeError.log".
make: *** [Makefile:1622: cmake_check_build_system] Error 1
Invoking "make cmake_check_build_system" failed

Thanks,
Ranjit

ranjitkathiriya, 2020-11-30:
Solved!! Thanks

closed 2020-11-30 by ranjitkathiriya

gavanderhoorn, 2020-11-30:
To help future users with potentially similar problems, it would be good to mention how you solved your problem.

ranjitkathiriya, 2020-11-30:
Just clean catkin workspace 

https://answers.ros.org/question/66366/how-to-catkin_make-clean-just-one-package/