ros-industrial/kuka_experimental #176: (KR-10 or KR-60) ROS Melodic on ubuntu 18/04

MBaranPeker, 2020-12-01:
Hi everybody, 
I am quite new with KUKA. Is there any example explain with details how to manipulate (KR-10 or KR-60) with ROS Melodic.

gavanderhoorn, 2020-12-01:
I don't know of anything else but [kuka_eki_hw_interface/krl/README.md](https://github.com/ros-industrial/kuka_experimental/blob/984e1f214246ab7631d1727f5428b0c815fe3a34/kuka_eki_hw_interface/krl/README.md) and [kuka_rsi_hw_interface/krl/KR_C4/README.md](https://github.com/ros-industrial/kuka_experimental/blob/984e1f214246ab7631d1727f5428b0c815fe3a34/kuka_rsi_hw_interface/krl/KR_C4/README.md).

Those only discuss configuration and some use of those packages though.

If "quite new with KUKA" means: I don't have *any* experience with KUKA robots (even without ROS), then I would recommend you register for basic KUKA training.

The packages and documentation (as little as there is) in this repository cannot be considered a substitute for proper training.

gavanderhoorn, 2020-12-01:
As this is not an issue with the packages in this repository, I'm going to close it.

Feel free to keep commenting on it of course.

closed 2020-12-01 by gavanderhoorn

MBaranPeker, 2020-12-01:
Thank you @gavanderhoorn. I am sorry for nobb question what is the difference between EKI and RSI ?

gavanderhoorn, 2020-12-01:
EKI and RSI are both external interfaces which allow you to communicate with KUKA controllers.

RSI is a hard real-time capable interface over UDP, which typically provides a bandwidth of about 250 Hz.

EKI is a similar interface, but runs at a slower rate and is less time critical.

Which of the two makes sense for an application would come down to what the requirements are.

KUKA has extensive documentation available on both EKI and RSI.

MBaranPeker, 2020-12-01:
Thank you so much for your perfect answer @gavanderhoorn