ros-industrial/kuka_experimental #177: Can't select  kuka_eki_hw_interface.src

MBaranPeker, 2020-12-03:
Hi,
I have some problem with kuka_eki_hw_interface.src file when i copy that file inside the controller. 
as described " Copy kuka_eki_hw_interface.dat and kuka_eki_hw_interface.src files to KRC:\R1\Program. " @BrettHemes  unfortunatly i can not select the kuka_eki_hw_interface.src file in hmi. It does not see it as a program. 
Have anybody experienced this before? 

thanks
[image]

gavanderhoorn, 2020-12-03:
It is [this file](https://github.com/ros-industrial/kuka_experimental/blob/984e1f214246ab7631d1727f5428b0c815fe3a34/kuka_eki_hw_interface/krl/kuka_eki_hw_interface.src) you've copied, correct?

MBaranPeker, 2020-12-03:
Yes it is

destogl, 2020-12-03:
I had a similar issue with the RSI-program. Try to copy it in some other folder and then move it using teach-pendant into the programs folder.

Or try to edit it with teach pendant. You have to convince KUKA's controller to "compile" it.

(This is how I remember, it was few years ago.)

MBaranPeker, 2020-12-03:
Thank you so much @destogl i will try.
Btw I confused about rsi and eki. Do u suggest rsi? Is it working stable?

destogl, 2020-12-03:
Regarding your issues this does not matter. RSI is working fine for even without RT-patch. We use a separate Network fo communication with the robot.

MBaranPeker, 2020-12-04:
@destogl thank you for your information

MBaranPeker, 2020-12-04:
hello guys i found a solution for that. you need to create new program with the name kuka_eki_hw_interface on workvisual and intall that program to controller then inside the hmi you can past  the code inside  kuka_eki_hw_interface.src and kuka_eki_hw_interface.dat file.

closed 2020-12-04 by MBaranPeker