ros-industrial/kuka_experimental #179: Eki Interface

MBaranPeker, 2020-12-08:
Hi,

I followed all the steps and read the issues in ROS Answers already. I found someone already have similiar error https://answers.ros.org/question/334974/failed-to-read-from-robot-eki-server/?answer=367184#post-id-367184

Already Available ports. Do I need to add new port 54600 ?

[image]

And I have issue with position_trajectory_controller as well how to solve it.
[image]

MBaranPeker, 2020-12-08:
When I check wireshark it looks kuka can not send to ros computer. 172.31.1147 is KUKA Computer. 172.31.1.5 ros computer    What could be the problem, any idea? 
    [image]

gavanderhoorn, 2020-12-08:
To be honest I have no idea right now.

My experience with EKI is rather limited.

If this is a problem at the networking level, I would suggest you reach out to KUKA, and ask them to assist you with configuring EKI on your robot.

This doesn't appear to be ROS specific, or specific to the component you're trying to use.

> Already Available ports. Do I need to add new port 54600 ?

If I understand the ROS Answers Q&A you linked correctly, the poster there mentioned things started to work when he *removed* the mapping for port `54600`, so if you haven't added it yet, then no, you should probably not add it.

gavanderhoorn, 2020-12-08:
Also: please do not post screenshots of terminals. There is no need, as it's all text.

Please copy-paste the terminal text into a comment instead.

MBaranPeker, 2020-12-08:
Thanks.

> Please copy-paste the terminal text into a comment instead.
ok

MBaranPeker, 2020-12-11:
Hi @gavanderhoorn came with update.

I got connection with 54601 port. In the link below shows data transfer between ros computer and Controller. 
[image]

Now we have another problem.
I open 2 teminals one of them for  launching the " test_hardware_interface.launch " file another one for 
running the " rqt_joint_trajectory_controller ". 
The problem here is while " test_hardware_interface.launch " running  " rqt_joint_trajectory_controller " does not start.

after getting the error

[ERROR] [1607657943.981633936]: Failed to read from robot EKI server within alloted time of 5 seconds.  Make sure eki_hw_interface is running on the robot controller and all configurations are correct.

" test_hardware_interface.launch " stops and " rqt_joint_trajectory_controller " starts.

What could be the problem here?
Why do i still get error even while connecting?

gavanderhoorn, 2020-12-11:
Is your controller in AUTO mode? Or is it in manual and do you start the program and then let go of the smartpad?

If the latter: that won't work and will lead to timeouts on the ROS side.

MBaranPeker, 2020-12-11:
It is working in AUTO mode. After trying many times I realize they transfer data randomly between up to 10. (As you can see above they have totally 6 transfer.but it is sometimes 8 sometimes 2 sometimes 3). I also think it is timeout problem. I could not solve that. 
Do you think Is it possisle to use TCP instead of UDP ? I tried but could not be successfull. I need to check with more detail. 
Do you know how to solve timeout problem?

Thanks

gavanderhoorn, 2020-12-11:
This still sounds like a networking problem.

I don't have any direct experience with EKI, so unfortunately I can't help you.

Perhaps someone else who monitors this issue tracker can help.

Random thought: have you checked whether you have a firewall active?

MBaranPeker, 2020-12-14:
Thank you @gavanderhoorn I deactivate the firewall. 
There were some backworker working in KUKA which i did not know. When we close them all everything worked well. Thank you so much again.

closed 2020-12-14 by MBaranPeker