ros-industrial/kuka_experimental #180: Guided needed for starters

ranjitkathiriya, 2020-12-21:
Hello Gavander,

I am new to Kuka robotic and I need help or links regarding the integration of my real Kuka robot with ROS.

Thanks,
Ranjit

gavanderhoorn, 2020-12-21:
Please don't call me Gavander, that's not my name.

> I am new to Kuka robotic 

As to your question: we cannot have a situation where `kuka_experimental` is your first experience with KUKA robots.

All of these robots are potentially highly dangerous, and to be able to use them safely, you *must* receive the proper training. At the very least the introductory training, which teaches you about safety and how to use your robot and controller normally.

>  I need help or links regarding the integration of my real Kuka robot with ROS.

We don't have any material specifically geared towards people without any experience with KUKA robots. I would say that is on purpose, for the above mentioned reasons.

Please make sure to get acqainted with KUKA robots and controllers first. Once you've done that, you can attempt to use the packages in this repository.

gavanderhoorn, 2020-12-21:
As this is not an issue which reports a problem with any of the packages in this repository, I'm going to close it.

Feel free to keep commenting on it of course.

closed 2020-12-21 by gavanderhoorn